iNavFlight/inav #2599: Failsafe RTH -> switch to drop when inverted

DzikuVx, 2017-12-17:
There should be a additional check if Failsafe RTH that detects the state when copter is upside-down.
It happened today for me: copter went inverted during landing, crashed and motors kept going until I was able to disconnect battery.

I'm not sure this is a feature request or a bug tho ;)

shellixyz, 2017-12-17:
This would help in that case: #2331

giacomo892, 2017-12-18:
Also maybe some other sanity check can be placed. Like if during automatic landing the copter banks more lets say than 90 degs then cut motors.

DzikuVx, 2018-02-03:
another observed condition: should fallback to drop when cell voltage below sane values

ErnestFPV, 2018-02-03:
Happened to me today, 4s pack dropped below 9v, quad crash landed into woods far away from me, motors were spinning till pack completely died (around 5mins) draining all from pack so i couldnt even use beeper to find quad when i arrived there. So it shouldnt be based on degrees but on pack health compared to cell number.

closed 2018-05-27 by stale[bot]