iNavFlight/inav #2600: Omnibus F7 Pro not detecting on iNav 1.8.1

Gaz93, 2017-12-17:
I'm having issues connecting to my board after flashing to the latest 1.8 iNav firmware and tried 1.7.3 but none give me a successful connect. I have no issues flashing the board in DFU mode and flashing it back to Betaflight lets me connect.

I've tried using zadig and reinstalling virtual drivers but when flashing the board is fine.

"Omnibus F7 Pro Flight Controller Built-in Dual Gyro AIO OSD Current Sensor and LC Power Filter"

https://www.banggood.com/Omnibus-F7-Pro-Flight-Controller-Built-in-Dual-Gyro-AIO-OSD-Current-Sensor-and-LC-Power-Filter-p-1169518.html?rmmds=myorder&cur_warehouse=CN

Using configurator 1.8.1 and firmware version OMNIBUSF7 1.8

shellixyz, 2017-12-18:
I could be wrong but it looks like it isn't supported. Did you try the V1 and V2 versions ?

Gaz93, 2018-01-03:
I've tried the V2 version too same result, I think I'll just try a different board for the moment.

closed 2018-01-03 by Gaz93

putimir, 2018-01-16:
Same here: Airbot Omnibus F7 v2 (supposedly supported). I can connect to the board flashed with betaflight, but when using Inav 1.8.0, I get 2 instances of "STM32 Virtual ComPort in FS mode", in the USB devices but no COM port.

Anyone figured this out?

putimir, 2018-01-16:
Hi,  #2647 fixed the duplicated STM devices, and I can now connect to the board...

Gaz93, 2018-01-16:
I had noticed the board started working after trying the newest release and I can say I'm a happy fellow now so I closed the issue :)

teknokraze, 2018-03-29:
For anyone else coming late to the party like me, the quick and easy fix is to use ImpulseRC's driver fixer downloaded from here:

[ImpulseRC Driver Fixer](https://impulserc.com/pages/downloads)

Solved it for me.