iNavFlight/inav #2602: Compass shows 0 all the time on Radiolink SE100 (M8N + HMC5983)

andrejpodzimek, 2017-12-18:
On my [`OMNIBUSF7V2`](http://www.readytoflyquads.com/flip-32-f7-omnibus), the compass is detected on iNav, but shows just zeros all the time (on the sensors tab). (It won't get detected at all on Betaflight.) [As discussed in this bug report](https://github.com/betaflight/betaflight/issues/4531#issuecomment-352309031), the device may have some compatibility issues. I've tried to make it work with both the iNav Git master and with release 1.8. The compass still shows zeros on all the three axes, despite calibration attempts. (All of `magzero_{x,y,z}` also stay at 0, so the calibration is failing.) Setting `i2c_speed` to `200kHz` didn't help either.

    # status
    System Uptime: 76 seconds
    Current Time: 2017-12-18T02:10:34.237+00:00
    Voltage: 5 * 0.1V (1S battery - NOT PRESENT)
    CPU Clock=216MHz, GYRO=MPU6000, ACC=MPU6000, BARO=BMP280, MAG=HMC5883
    STM32 system clocks:
      SYSCLK = 216 MHz
      HCLK   = 216 MHz
      PCLK1  = 54 MHz
      PCLK2  = 108 MHz
    Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
    SD card: None inserted
    Stack size: 6144, Stack address: 0x20010000
    I2C Errors: 1, config size: 3056, max available config: 32768
    ADC channel usage:
       BATTERY : configured = ADC 2, used = ADC 2
          RSSI : configured = ADC 3, used = ADC 3
       CURRENT : configured = ADC 1, used = none
      AIRSPEED : configured = none, used = none
    System load: 7, cycle time: 254, PID rate: 3937, RX rate: 112, System rate: 9

andrejpodzimek, 2017-12-18:
Ah, now I see that this is a duplicate of #2299. OK, whatever. If someone wants me to experiment and/or test some changes, I'm happy to help.

closed 2017-12-18 by andrejpodzimek

alanfj, 2018-01-19:
Is there going to be a fix for this(Radiolink SE100 GPS Compass)compass issue, in future releases?

digitalentity, 2018-01-21:
@alanfj definitely we would. My SE100 GPS is a few days away from me. I'll debug and fix as soon as I get it.

alanfj, 2018-01-22:
Excellent!, You are the "Man!" This is a really nice GPS/Compass hardware, w/ a great GPS and really cool LED"s, which will be a great addition to the INAV, Betaflight, Cleanflight, hardware library addition!. Thanks again, and way to go digitalentity!  Alan Farms Jr.