iNavFlight/inav #2603: Help - OMNIBUS F4 v2 Pro Servo big Delay

Christianmix, 2017-12-18:
Due to my bad English, I made a video without speaking, just showing what happens.
I'll make it well explained here, just look at the video that will see that there is a delay in the servo movement, even when reaching the end with the transmitter stick, the servo moves slowly, this in Angle activated mode, when leaving the inav in passthru this does not happen, but when returning to the angle mode it does again, when I lift the accelerator stick and it worsens, in addition to slowly go to the position that I place the stick, when releasing the stick the servo gradually returns to the normal position look first here he going to wander and come back even after having released the stick https://youtu.be/q2SgRFqvyGg?t=321 and then in this other moment of the video, I show that the servo comes back quickly but well after having the transmitter stick ... https://youtu.be/q2SgRFqvyGg?t=417 .............. I do not know what to do anymore I tried on 2 boards f4v2 and 1 cc3d and gave the same problem, when the plate is moved the servo works quickly, without any delay, the delay in servo response only exists through the transmitter, but with the card connected to the usb and moving the stick of the transmitter can see in the program that the commands come quickly on the board look at this moment ... https://youtu.be/q2SgRFqvyGg?t= 50
please help me, I do not know what to do, I've tried several versions of inav and all gave the same problem, I must be doing something wrong and I can not find out.
Remembering that I have already tried on 2 other boards besides this one and with 6 other servos, it must be some configuration error.

complete video: https://www.youtube.com/watch?v=q2SgRFqvyGg

shellixyz, 2017-12-18:
Please post your config dump

shellixyz, 2017-12-18:
To me it looks like your servo is struggling. Are you sure it is powered appropriately and that it is working ? Did you try with another servo ? If the issue comes from the servo it would explain why it didn't work on two different boards. Other than that quickly looking at the settings you showed I can't see what could cause that.

Christianmix, 2017-12-18:
#shellixyz, when this in passthru works perfectly, the problem only happens when it is in some way stabilized or failsafe. This is the dump.

# dump

# version
# INAV/OMNIBUSF4PRO 1.7.2 Jul 20 2017 / 11:59:41 (81b85de)

# resources

# mixer
mixer FLYING_WING

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature RX_SERIAL
feature MOTOR_STOP
feature GPS
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
[... 454 lines cut ...]
set nav_mc_pos_z_d = 0
set nav_mc_vel_z_p = 100
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 35
set pitch_rate = 9
set yaw_rate = 20
set tpa_rate = 33
set tpa_breakpoint = 1300

shellixyz, 2017-12-18:
Just some ideas

I think you should avoid using 90° as max roll/pitch, 45° should be safer to start with
```
set max_angle_inclination_rll = 900
set max_angle_inclination_pit = 900
```

Also please test in acro mode (no mode configured) first with
```
set fw_p_pitch = 0
set fw_i_pitch = 0
set fw_p_roll = 0
set fw_i_roll = 0
```

Christianmix, 2017-12-18:
I already fly a long time fixed wing, with a cc3d, I put 90º pr that I do not like long corners, but I discovered one thing, I feed my servo through my esc, off the board, but when I put it on the board does not work . I checked the pins where I connect the servos, before connecting the servo all the pins are with 4.7v, when connecting only 1 servo, that voltage drops to 1.06v on all pins, less on the SBUS, on the pin that gives the current to the receiver still 4.7v, will the 2 new cards have problems?

shellixyz, 2017-12-18:
I'm not sure to understand but if your 5V rail goes down to 1.06V it can't work that's for sure. Make sure your 5V rail can feed enough power to your servos and also that your servo grounds are connected directly to your FC. Make sure your servo power rail is independent from the FC 5V rail.

It is unlikely that all your boards have the same issue.

shellixyz, 2017-12-18:
Where are you from @Christianmix ?

Christianmix, 2017-12-18:
Brazil

Christianmix, 2017-12-18:
I use cc3d with inav and never isui the board to power the servos, always use the bec of the esc, but that would not be problem to give that delay so great in the servo, if the problem was in the servo when in the passthru mode should happen same problem, I think it's the same board, it does not have to stay this way, the answer is very slow when using some flight mode, I tried with the air mode and also happened the same.

Christianmix, 2017-12-18:
look how the voltage drops

https://youtu.be/X53EXUvqfZY

shellixyz, 2017-12-18:
From the video it looks like you are powering the board from your ESC BEC 5V. Please check from your FC documentation but from what I know the F4Pro board has an onboard "BEC" and should be powered directly from your battery. If the FC's output power rail get 5V from the FC 5V you should only connect your servo ground and signal to the FC. The 5V wire for the servos should not be connected to the board but directly to your ESC BEC 5V.

You can quickly confirm if the issue is not coming from the software by disconnecting everything from your FC but keeping the receiver. Then connect the FC with USB to your computer and external power if it is needed for your receiver. Finally connect to your FC with the configurator and go to the motors tab. There you can see what the software is commanding the servos to do. If the output values are moving as expected then it is an hardware issue.

Christianmix, 2017-12-19:
Hi I did the tests as you said, with nothing on the board, and I looked at the engine flap as the servos behaved and to my surprise they appear very problematic, but with the same PLCA in the beta flight is perfect, and to despair I took another f4 v2 hehehehehe are 3 now in front of me, if I put in some mode of flight, I have problem with the servos, no flight modes is perfect, but I want the flight stabilized, so I did not advance the controllers anything , I did a lot of research on the internet and did not find anything, it follows the video with the demonstration of how the graphics of the servants behave in the programs BF and Inav. Thanks for the help, but I'm thinking of giving up and going back to old apm that never left me in the hand.
 video

https://youtu.be/5FUp_Qca0Bs

shellixyz, 2017-12-19:
What mode is enabled in your last video ? Also did you set roll/pitch P and I term to 0 ?

Christianmix, 2017-12-19:
when p i d this zero works well. because?
same mode the last video

look

https://www.youtube.com/watch?v=hdv05oGqKZ4

shellixyz, 2017-12-19:
It is because when your board is armed the I term accumulates the error of the gyroscope. When on the ground you should only test with the board disarmed. When the board is disarmed the I term have no effect.

Christianmix, 2017-12-19:
so the sign is good and this error will only hit the ground? When you're gone, will not it be okay then? can I put the default values of P I?

shellixyz, 2017-12-19:
For your first flight your P and I terms should be 0. Always test your aircraft first in passthrough and carefully test the other mode(s) while being prepared to switch back to passthrough in case there is a problem.

For tuning the PID controller please follow this guide:
https://github.com/iNavFlight/inav/wiki/Tune-INAV-PIFF-controller-for-fixedwing

Christianmix, 2017-12-19:
I understood, I now tested with horizon mode without being armed and with the default pid and it really did not give any problem, so the one who was creating the problem was just me activating the controller on the ground, but one thing happens that I found strange, if I hold the stick trtansmissor in any maximum position, when lifting the stick of the accelerator, even if I do not decrease the commands of the other stick they will decrease according to how much increase the accelerator, as if it were a mix of the plate making that the more speed, less command . is this right?

shellixyz, 2017-12-19:
Yes this is all right this is because of your tpa_rate setting

shellixyz, 2017-12-19:
With increased air speed (inferred from increased throttle) your control surfaces have more control so the movement should be less for the same effect.

Christianmix, 2017-12-19:
ok thank you very much, you helped me a lot in the mistake that I was causing, thank you for the patience you had with me.
now I just have to find out why I can not power the servos directly on the board, none of them I can, the only connection that stays constant the 5v is in the sbus, in the servo connections when I put the servo the low voltage and they do not work, it will be do I have to activate anything on board or inav?

shellixyz, 2017-12-19:
You should just have to split the red wire from your servo connectors and connect them directly to your ESC BEC 5V.

Christianmix, 2017-12-19:
is really this is the best solution, i bought 2 more boards f3 10 dof to use, i am hoping to get from china, i prefer the connection mode of it, in these f4 i am using op gps and i had to adapt the bigger connector because the port where this original gps connector comes in, it's uart 3 which is the port that sbus uses, I just thought it was wrong, inav should give us the option to put the uart according to our need, that is, I could have a programming where we would place the connector that is in the same uart of sbus in another uart, I do not know if you understand me? my gps is from the small 4-wire connectors and the connector for it on f4 v2 is in the same uart as SBUS, this should have change, type a programming where i can put the SBUS on uart 6 that i do not use the connector, would much easier than having to go behind a large connector and use only the first 4 wires.

shellixyz, 2017-12-19:
F4 boards don't have inverters in the MCU that's why you can connect SBUS only to one of the UART. The one equipped with an external inverter.

shellixyz, 2017-12-19:
F3 board have integrated inverters in the MCU allowing to use SBUS on every one of the UART units.

Christianmix, 2017-12-19:
I wanted to know half of what you know, it would be great for me, did you learn all this by using the inav or did you take a course? if it was only using the inav should be time that isa it, I met the inav because I had a cc3d here and not used for lack of rth, and since I always fly more than 3km I can not take the risk of losing radio signal or video and not having an automatic return home, it was when I read about inav, watched some videos of Matt and I configured my cc3d, it was perfect, then I decided to buy the f4 to put in a mini heel, but I could not understand how to do the mixagen of the tail then discarded the project, I thought I'd ask for a copy of Matt's configuration, but I was ashamed. I'll wait and buy a 2.8 APM that I know set up for the tail tail. But I thank you so much for helping me. thank you

Christianmix, 2017-12-19:
Mini talon v Tail

Christianmix, 2017-12-19:
, then I decided to buy the f4 to put in a mini talon v tail, but I could not understand how to do the mixagen of the v tail then discarded the projec

shellixyz, 2017-12-19:
I'm a software engineer so it is helping to understand the iNav software but other than that all I know about about flight controllers and how to use them comes from the internet. I read all the free flight controller software documentations (iNav/Betaflight/Cleanflight/LibrePilot/Ardupilot). Also there is a lot of good videos on youtube: [Painless360](https://www.youtube.com/user/Painless360) and [Project Blue Falcon](https://www.youtube.com/channel/UCObMtTKitupRxbYHLlwHE3w) for example.

For iNav the best is to start with the [wiki](https://github.com/iNavFlight/inav/wiki) and the [docs](https://github.com/iNavFlight/inav/tree/master/docs).

To configure the iNav mixer for a Mini Talon V tail you should be able to use ready to use config:
[Custom mixes for exotic setups - V tail fixed-wing](https://github.com/iNavFlight/inav/wiki/Custom-mixes-for-exotic-setups#v-tail-fixed-wing)

For an explanations please watch this video by Painless360:
[Setting up custom servo mixes in iNav for V tail](https://www.youtube.com/watch?v=qL2LMFF5bJ4)

PScott1, 2017-12-20:
@shellixyz  JC's  (Project Blue Falcon) died in a motorcycle accident a few months ago...his videos are still great.

shellixyz, 2017-12-20:
@PScott1 I know. Very sad !

Christianmix, 2017-12-20:
I was thinking here about the accelerator changing the commands, is that why when I accelerate my fixed wing starts losing altitude?

shellixyz, 2017-12-20:
Probably not.

Are you talking about Angle or Horizon mode ? If yes then it is probably because the way you calibrated the accelerometer makes your model fly flat relative to the ground with the pitch command centered. You need to increase `pitch degrees` in the configuration tab until your aircraft is not gaining nor losing altitude when your pitch command is centered.

It can also be a thrust angle issue. What fixed wing model are you talking about ?

Christianmix, 2017-12-20:
i talking about fly wing.

Christianmix, 2017-12-26:
Hi, I flew today, and until it went well, until at some point the fly wing started to fly badly and in osd gave message that it was necessary the calibration, I do not know why this, but this is not the biggest problem, the biggest Problem was the flick of the osd when I accelerate, when I decrease the throttle back to the norm, I'm using the same battery for everything, then the battery negative then the camera and the video transmitter, it's the same negative that also turns on the flight controller . here the black wire is negative and the red wire is positive. In the flight controller only the video signal is on. Look at the video as it was. 

https://youtu.be/N-k6R7KdWlw

digitalentity, 2017-12-26:
@Christianmix you have too much noise on your camera/VTX power. Try powering them with a separate small battery and if it works - invest in a good LC-filter.

closed 2018-05-27 by stale[bot]