iNavFlight/inav #2605: Oscillation detection and avoidance (safety)

krzysztofmatula, 2017-12-18:
Some time ago I was experimenting with various presets found in the iNav configurator and some of them (those for bigger machines) made my 280mm quad oscillating so hard that even with zero throttle it was going up quite fast and the only way to bring it down was to disarm in the air...

It was obviously a tuning issue, but I wonder if some extra safety feature could be added here?

Maybe heavy oscillations could be auto-detected and some adjustments could be automatically applied? Something like TPA?

Or maybe something that would ensure the machine is not going up when zero throttle is commanded?

krzysztofmatula, 2017-12-18:
I think the presets are just from someone's setups.
The ones for 10" and 12" are *way* too aggressive for my big machine as well.

giacomo892, 2017-12-19:
Oh my F450 with 2212 980Kv motor and balanced DJI 9450 propellers I ended up with P=65 and D=45 on both pitch and roll axis and it is behaving perfectly if this helps. Then I needed some filtering since my FC was even hard mounted. Like software LPF to 50Hz and hardware LPF to 98Hz. No notch applied since it is an F1 machine.

Dronek, 2017-12-19:
Oszillations also occur on fixed wings, especially with current implementation of TPA when diving with zero throttle. I think this is going to be addressed together with improved airspeed functionality. However, a general detection of obvious oscillation might be a good safety feature, additionally. I guess there are several scenarios where detrimental / resonant behaviour of a quad or fixed wing could be detected and avoided.

sammichael25, 2018-01-02:
My f450 does exactly this
https://www.youtube.com/watch?v=tKcMasD6HX4

I can’t seem to get the issue fixed

krzysztofmatula, 2018-01-02:
That's the issue to be addressed.
I guess reducing P and D for pitch and roll (e.g. by dividing them by 2) would help and that's the thing that could be done automatically, allowing to get the machine safely down.

closed 2018-05-27 by stale[bot]