iNavFlight/inav #2607: STM LIS3MDL compass support ?

fiveangle, 2017-12-20:
### LIS3MDL Compass Support ?

In looking for an ultra-lightweight GPS-Compass module, and an RCG member suggested this:

https://store.mrobotics.io/product-p/mro-ugps-samm8q-01.htm

But noticed that it uses the STM LIS3MDL compass:

http://www.st.com/en/mems-and-sensors/lis3mdl.html

 [Reviewing here](https://github.com/iNavFlight/inav/tree/development/src/main/drivers/compass), it looks like this isn't directly supported.

Does anyone know if this uses the same protocol as one of the supported magnetometers and so should work ?  Or would support have to be fully ported ?  I did a bit of digging and suspect it isn't a drop-in as PX4 support looks like it was added March 2016 and looks to be an entirely separate driver:

https://github.com/PX4/Firmware/blob/master/src/drivers/lis3mdl/lis3mdl.cpp

The reason for asking is that I'm trying to implement the lightest-weight solution possible for emergency position-hold (and *maybe* RTH if something goes wrong with video feed).  It will otherwise be dead weight on every flight, and at <8g, I can't find anything close to this with an M8.

This is probably the alternative if I can't find an M8 unit - this is <10g NEO-6M:

https://hobbyking.com/en_us/micro-hkpilot-gps-and-compass-u-blox-neo-6-and-hmc5883.html

Thanks,

-=dave

digitalentity, 2018-02-21:
This issue somehow didn't grab my attention. LIS3MDL is not supported by INAV and will require a separate driver to be written. I don't have the hardware, so I can't implement & test.

fiveangle, 2018-02-21:
Thanks for confirming.

The LIS3MDL is about [2 bucks at the usual suspects](https://www.aliexpress.com/wholesale?catId=0&initiative_id=SB_20180221082345&SearchText=LIS3MDL) so I could order a couple and send you one if/when you might get around to it 🤷‍♂️

digitalentity, 2018-02-22:
Purchased a few.

abenn1, 2018-06-09:
I'm new to the way things work here.  Does the last comment mean that LIS3MDL support will be in iNav v2.0?  My favoured supplier (Drotek, France) now uses it in its M8N modules.

fiveangle, 2018-06-10:
It means the target goal for implementing support for LIS3MDL is 2.0.  That doesn't necessarily mean it will be in that version ;)

abenn1, 2018-06-10:
Thank you fiveangle.

bforbort, 2018-07-31:
Sorry folks, I didn't realize this was a requested feature.  I actually wrote a driver for the LIS3MDL for v1.9.1 a while ago.  See d7fedd61030a4b73d6f1307e2a0d24120fef7915

I'll see if I can rebase it and do a PR.

digitalentity, 2018-07-31:
@bforbort cool, please post a PR

fiveangle, 2018-09-20:
@bforbort - It seems you completed your port to BF (https://github.com/betaflight/betaflight/pull/6514) let us know if you're interested in rebasing / submitting pr.

Thanks !

-=dave

fiveangle, 2018-11-22:
@bforbort. Curious if this ever got sorted out ?

shellixyz, 2018-11-22:
@bforbort  Support has been merged August first.

@digitalentity Looks like we can close this ?

digitalentity, 2018-11-22:
Yep, I think it could be closed.

closed 2018-11-22 by digitalentity