iNavFlight/inav #2609: [FEATURE REQUEST] Use GPS Alt if there's no baro in the OSD

wx4cb, 2017-12-21:
Have a way of having the GPS altitude in the OSD instead if there's no baro onboard.

Same discussion here on the betaflight github: https://github.com/betaflight/betaflight/issues/4800

Pairan, 2017-12-22:
I don't know if I get this Feature Request right, but I fly without Barometer and in fact have ALT in my OSD which only can be measure by the GPS which is installed!

In case you intended to archieve this ... ehm ... its done ! :D

wx4cb, 2017-12-22:
hmm.... strange. i don't get altitude on my osd (unless im missing something) can you post a pic of your osd config screen. on mine all i get that says "gps" is speed heading lat/long HDOP course etc.

The only "altitude" one is checked, but i don't see a GPS Altitude

krzysztofmatula, 2017-12-22:
scarab-osd has two separate altitude widgets:
- "Altitude - shows the altitude from prime altitude source BARO is used else GPS"
- "GPS altitude - display GPS altitude"

I wonder if we'd also prefer to have both in iNav - one showing the altitude FC uses for computation, and the second - raw reading from GPS to give pilot another view...?

ShawnzAZ, 2017-12-29:
I'd like to have the option for Altitude MSL vs just AGL from launch.

wx4cb, 2017-12-30:
@krzysztofmatula the issue with that is, they COULD be both the same if there's no baro. that was my reasoning behind, (in the betaflight thread i linked to) just leaving "altitude" in the current spot, and then adding the gps alt code into where it does all the computation and if there's a baro, use that if there's no baro then use the gps alt if it's there.

cj7hawk, 2018-04-15:
Did anything happen with this? It doesn't appear to be present in 3.3.1 - Will it be included in 3.4?

DzikuVx, 2018-04-25:
INAV Altitude comes from position estimator. It merges all available sensors.

closed 2018-04-25 by DzikuVx

Riffer, 2018-10-24:
@DzikuVx does this mean Altitude in OSD and Telemetry is coming from both Baro and GPS and the values are calculated together? If yes, is the mixup in the calculation of Altitude influenced by nav_w_z_baro_p and nav_w_z_gps_p? If not: Where can one set 'show altitude from [baro|gps] only?

DzikuVx, 2018-10-24:
In "sensors" tab in configurator or in blackbox log