iNavFlight/inav #2614: Home resets during FS cycling

Hwurzburg, 2017-12-22:
This is a potential fatal issue!

_Omnibus Pro F3
_INAV/OMNIBUS 1.8.0 Nov  1 2017 / 06:14:28 (912d1315)

Problem:
Home point resets while going from in FS mode to signal re-acquisition.
First tests of flying to FS point while hands off in ALTHOLD+STABLE mode in flying wing. FS is entered and RTH begins normally....as it turned back toward home in RTH the OSD flashed "ARMED" briefly, and then the FC re-entered FS, but home point is now at current distant location...luckily, by re-orienting TX signal was re-acquired and plane recovered.

the plane operates in all nav modes properly. FS is setup in RX and FC for no pulses using XM+ rx.
The only "non-standard" setup items I can think of is that it is using an ARM switch on a high channel (ch 11) and that I had fixed_wing_auto_arm enabled from previous uses which did not have an ARM switch configured. I suspect that the recent code fix for this feature in the arming sequence may be the culprit. Perhaps when an arm switch is used the FS code requires that this feature be disabled, although it is not stated?

DIFF is here: https://pastebin.com/JVHks7Qk

digitalentity, 2017-12-23:
Something caused machine to disarm and re-arm in the air. Do you have logs by any chance?

Hwurzburg, 2017-12-23:
no, sorry....am a little afraid of re-attempting with an SD card in to get them with auto_arm enabled, which my gut says is somehow the issue, since the arming code was changed to make it work nominally....but when weather cooperates after XMAS I will attempt it again with SD card logging...

Hwurzburg, 2017-12-30:
I have flown again several times, toggling the "RANGE" check on my taranis to invoke toggling FS's.....so far, I have not seen another flash of the arming screen and reset of home position....I will keep doing this every time I fly to see if I can capture it with logging on, but at this point its a one-off gremlin...

digitalentity, 2017-12-31:
I guess we can close this. Failsafe by itself doesn't cause home reset.

closed 2017-12-31 by digitalentity