iNavFlight/inav #2623: A983 compass use

Xgdc, 2017-12-27:
I have a M8N GPS with onboard compass A983. I've tried all the existing options in the configurator but it remains undetected (I have another 6M GPS with onboard compass L883 and that works) this using an Omnibus F4 pro V2.1 board. Is there any way I could make the A983 compass work as I would like to use the more advanced M8N GPS? I am using: inav_1.8.0_OMNIBUSF4V3.hex

digitalentity, 2017-12-28:
Which GPS module is this? Can you post a picture of the module and where did you purchase it?

Xgdc, 2017-12-28:
[image]

I hope the image is visible, It was bought (some time ago) from Good Luck Buy and is called: NEO-M8N Flight Controller GPS Module with Shell & Stand for PX4 Pixhawk V2.4.6 APM 2.6 Compatible.

digitalentity, 2017-12-28:
Can you send a link to a product? I'll purchase one. This is likely another instance of cloned HMC5883 (or pin-to-pin compatible knock-off with different API)

Xgdc, 2017-12-28:
http://www.goodluckbuy.com/neo-m8n-flight-controller-gps-module-with-shell-and-stand-for-px4-pixhawk-v2-4-5-apm-2-6-compatible.html

As I said, this was bought some time ago but they do currently still offer it for puchase (I've just checked).

Xgdc, 2017-12-29:
So this is what i decided to do. Fitted a whole new L883 Mag board (filed to fit and hot glued) Works great!

[image]

closed 2017-12-29 by Xgdc