iNavFlight/inav #2626: Omnibus F4 V3 is compass Mag. required for RTH to function in fixedwing

trivejb, 2017-12-28:
I am installing the Omnibus F4 V3 in a fixed wing (EasyStar Clone) and have everything working except RTH.  STAB/PASSTHROUGH settings are properly displayed in the OSD but when switching to RTH it displays AIR. 
 The GPS I currently have does not have the compass built in (but have just ordered another from RTFQ with the compass). I am having a little trouble understanding if a compass/mag is required for RTH.  I so then the new GPS/compass should solve the problem.  If RTH can be enabled without the compass I will post screen prints, etc that may help someone help identify the problem. 
Thanks,

linuxras, 2017-12-28:
RTH does not require a compass on a fixed wing

trivejb, 2017-12-28:
Ok what I think may be happening is that RTH may not be able to be tested on the bench. All of the "Pre-arming checks" are green until I switch to RTH then the "Navigation is safe" turns red.  I have 8 satellites and a 3D fix. Only thing I can think of is my alt is at -3 feet or the FC knows the plane is not flying and will not go into RTH. Any ideas?

trivejb, 2017-12-28:
Found a post on RCGroups commenting that all sensors must be set to Default.  I do not understand this but I have a message "Reset Z axis and the value is greater that zero.  I understand that it must be zero you should look in flight direction from behind onto the tail of your model (you should see "BACK" on the screen?? Will try and get this corrected and see if that helps.

Mow06, 2017-12-29:
Read the wiki

nav_min_rth_distance | 500 | Minimum distance from homepoint when RTH can be activated

lancaster100, 2017-12-30:
i have installed the omnibus F4v2 board using F4v3 firmware in a plane
I can only trigger nav modes, eg navposhold when the compass is operating
is this right?

trivejb, 2017-12-31:
Actually the instructions for fixed wing recommend NOT to use the compass.
"If your compass is not 100% properly setup just disable it instead. A calibrated compass can cause orientation drift during flight that may not show up in the configurator (especially built-in ones on your FC). Really consider disabling it unless you need it. INAV uses GPS heading normally, Only on ground before GPS speed has been high enough or if error between GPS heading and compass heading exceed 60deg will it use compass heading"
Check out the instructions for fixed wing 
https://github.com/iNavFlight/inav/wiki/Fixed-wing-guide

lancaster100, 2017-12-31:
I flew the plane today with both GPS and compass working - the plane behaved totally as it should
But when I disable the compass on the bench, the nav modes stop operating
Any ideas?

trivejb, 2018-01-01:
I have not flown mine yet because of the weather…cold and windy.  I currently only have a GPS and just tried some of the nav modes and get the same result as you… none work.  On the tools tab all Pre-Flight markers are green including “safe for navigation”.  In the mode tab it shows the switch moving to the selected green area but the name on the left does not change color, to green I think.   When I toggle to RTH the OSD monitor shows PASS and I believe it should show RTH . So appears I may may have the same problem.  I have ordered a GPS/MAG and it should be here Tuesday so that will give me a chance to test it. Let me know if you find the answer.

trivejb, 2018-01-01:
Went back and read through the the airplane guide and I wonder if the phrase "INAV uses GPS heading normally, AND Only on the ground before GPS speed has been high enough.....it will use compass heading" could mean that when sitting on the bench/ground and GPS speed is zero then none of the inav modes can be made active UNLESS there is a compass??? Maybe someone else will help with this.

lancaster100, 2018-01-03:
trivejb
You might be right - perhaps on the bench the nav modes are inoperative due to lack of forward movement.
When the weather improves I will flight test my plane with nav modes set up plus GPS but without compass and report back

DzikuVx, 2018-01-03:
No, MAG is not required on fixed wings. Closing, thanks

closed 2018-01-03 by DzikuVx

trivejb, 2018-01-03:
Was watching this video on YouTube (https://youtu.be/2hH7NxOAKns)
 about the Omnibus F4 GPS wiring issues and posted a comment asking the same question. Below is a reply I received so this may be the way Nav modes function.
Thanks

Highlighted reply

"EnglishTurbines
13 hours ago (edited)
Hi Blackfox.....Actually, you will have to Arm AND launch the model before RTH will activate for you....I think there is also a minimum distance from Home before it will go active.
  What you describe is normal behavior.  Watch my channel on the Falcon to see a Demo of iNAV.﻿"

lancaster100, 2018-01-03:
trivejb
Thanks for the link - that's interesting
I know RTH will not operate when the plane is stationary, but the problem also is present with the other nav modes, so the procedure should be clarified because when you set them up in the tab you dont know if the FC has accepted them until you fly

lancaster100, 2018-01-13:
trivejb
I have flight tested the F4 V2.1 board (target F4V3) with and without the compass
All the modes including navmodes, RTH and Failsafe worked really well with no interference between sensors - so you were correct to assume that although the navmodes dont switch on without the compass enabled whilst bench testing, they operate ok in the air
It all makes sense now!