iNavFlight/inav #2631: Problem with SmartAudio on SoftSerial on OmnibusF4Pro

sergeysi779, 2017-12-29:
I'm trying to use SoftSerial for SmartAudio on OmnibusF4Pro target (knockoff Omnibus F4 board from AliExpress) and AKK X2 VTx. I've connected CH6 pad on board to SmartAudio wire on VTX and configured SoftSerial1 as TBS SmartAudio peripheral in iNavConfigurator. It doesn't work, in OSD it shows transmitted packages and no recieved ones. However it works on the same setup in Betaflight 3.2.3. I just configure SoftSerial with "resource serial_tx 11 C09", and it works. I am wondering whether C09 pin is the same as PC9 pin in iNav source. I also found an issue (https://github.com/iNavFlight/inav/pull/1576) where Soph also can't use SA on SoftSerial. What could be wrong?

sergeysi779, 2017-12-29:
SA also works on hardware UART in iNav with same setup. So looks like problem with SoftSerial.

shellixyz, 2018-01-15:
Softserial pins for OMNIBUSF4PRO` are `PC8` for RX and `PC9` for TX. C09 in Betaflight is the same as PC9 in iNav. I can't say more I never used softserial with iNav.

heady123, 2018-01-27:
I do have the same issue with an original Omnibus F4 Pro v3 and a TBS Unify Pro v2. CH6 (which is PC9 according to the datasheet) connected to the audio of the VTX. I don't have any spare UARTS cause i need them for GPS, Mag and RX. Do we need to set any specific feature or baud rate?

heady123, 2018-02-28:
time progressed and I tried a few things meanwhile. I tried to use a 3.3V to 5V Uart level converter to boost the uart levels to 5V, no luck. I really have no idea left and am waiting for access to a oszilloscope to further test this software serial. 
As I didn't get it to work I sacrificed my Compass and have it now running on a hardware uart which works fin from a smartaudio perspectve. however I hooked it up in parallel with a microphone and according to TBS http://team-blacksheep.freshdesk.com/support/solutions/articles/4000100084-tbs-smartaudio-what-it-is-and-how-it-works you should only hear the Smartaudio clicking once at startup and whenever you change channels. However inav seems to be sending the smartaudio signal over and over again which results in a constant clicking noise in the audio. Can this be disabled?

heady123, 2018-02-28:
[image]
[image]
NOw that my high end scope is built ;) I had a look at the signals coming from the different Serial Interfaces.
First image is the output of the softserial. The second image is the output of the Hardware based UART (UART6). 
To me it actually looks like this Softserial is inverted. could this be the case? i mean the solder pad is directly connected to the pin on the F4, but afaik softserials can be either initialized with a pull up or down, correct? Can we get a parameter for this, or something in the Board definition file to fix this. Will try to get a hold of an inverter and test it.

heady123, 2018-03-01:
Made 2 better images from the whole SA packages. Doesn't really look inverted, but is high for the most time. then right before the package it starts going low. missing pull-down resistor? Softserial first, second is Hardware UART:
[image]
[image]

heady123, 2018-03-01:
soldered in an inverter (which I still had from my Taranis mod), confirmed that the signal actually was inverted, hooked up a TBS Unify Pro. Again, no luck. So its not just inverted.
Anyway, hope those pictures help and someone with actual knowledge of the protocol (which I don't have the definition for) can make sense of whats going wrong.

drmrb, 2018-05-23:
Using inav 1.9.1 smart audio v2 through ch6 soft serial works on my Airbot Omnibus f4 pro v3 and Unify Pro HV Setup. Used 19k baud rate setting.

UnTraDe, 2019-11-28:
> soldered in an inverter (which I still had from my Taranis mod), confirmed that the signal actually was inverted, hooked up a TBS Unify Pro. Again, no luck. So its not just inverted.
> Anyway, hope [...]

As far as I know SmartAudio is not inverted, but it does release the the line while it is not transmitting, to allow the microphone to use it. So a working SmartAudio line should look "inverted" (the line is low while the vtx and the fc are not transmitting) but its actually just in tri-state. Maybe the SoftSerial is not releasing the line?

BTW, the specs can be found here: https://www.team-blacksheep.com/tbs_smartaudio_rev08.pdf

closed 2019-12-24 by DzikuVx