iNavFlight/inav #2632: WingWing preset, flew with only ARMED, plane is very Unresponsive -> CRASH

junwoo091400, 2017-12-30:
Board : Omnibus F4 V3
Version : INAV 1.8.0

INAV/OMNIBUSF4V3 1.8.0 Nov  1 2017 / 06:35:50 (912d1315)

### Behavior
* On the ground, Plane's control surfaces move enough, same amount as previous successful simple flight w/ only Receiver. But when it flies, without ANY mode activated, just ARMED. The control is very very sluggish(When I command Bank Left FULL, it banks left about half 1~2 seconds later...), and  about 20 seconds later, after a Long Turn, it went out of control....

* I used mostly Z84-WingWing Preset, reversed Left Wing Servo, and set the Modes according to my Receiver. What can possibly go wrong?

----------------------------------------------------------------------------------------------------------------

* I just ARMED, and flew the plane, with no Modes activated at all.

* It should fly like normal pass-through, as I checked the correct movements on the ground.

* It is very very non-responsive, almost out of control. Leading to a CRASH :(

DUMP : https://pastebin.com/iFmdkWWi

[LOG00008.TXT](https://github.com/iNavFlight/inav/files/1594272/LOG00008.TXT)

=> At about 1 minute 32 seconds, the plane crashes :(

digitalentity, 2017-12-30:
From your log I see that plane is barely responding to commands. Most likely your mechanical rates are too low. INAV expects full servo deflection to defect aileron to ~45deg.

junwoo091400, 2017-12-31:
@digitalentity 
Thanks, I changed the mechanical rates (slightly better).

But I found out that the problem was that 'RATE' mode (that should be the default, with only ARM and no modes activated right?) didn't deflect Servo enough.

So what happened was when I was on the ground, I was DISARMED, and I checked Servo deflections and went "Oh that is OK". But when it ARMED, the Servo deflections were much, much smaller, leading into a crash... (Normal : 14mm  =>  ARMED,noMode : 1~2mm)

I changed PASSTHRU to be default for ARM, and it worked much better than the last crash...

Do you know why 'Default flight mode (RATE)' doesn't deflect Servo as much as when it is in DISARM state? (like, really it barely moves :( )

Redshifft, 2017-12-31:
You might want to take the RC expo out if it's set at the default 70% first.
Test fly and trim in passthru  and add any expo if you need it. If it will not fly how you require it to change throws etc.. INAV can only work with the mechanical constraints you allow it as DE said. You can play with rate and level modes after that.
Don't  forget if no mode is set it will be in Acro (rate mode) I don't know what you mean re,default arm mode? and anything other than passthru will be heavily influenced by the flight controller sensors.

junwoo091400, 2017-12-31:
@Redshifft 
Yeap, I should try Pass-through much more today when the sun rises.
And the core reason why Acro wasn't working was because of it's 'PID/Level-strength' Settings. 

Now I probably know what to do after reading Fixed-Wing-Guide about 3 times.
1) Passthrough => check MAX Rates -> set in Configurator
2) Tune PIFF controller with 'Acro' mode (Angle maybe?)
3) Try Angle Mode and RTH and so on...

giacomo892, 2017-12-31:
Just to be sure I set my Wing Wing setup correctly. How much deflection (in degs) should I get when I ask for max roll and pitch all together? (in passthrough of course). Maybe it is here that we may fail.

Redshifft, 2018-01-01:
@giacomo892 For a flying wing you are best to have more travel available for the test flight :)  You can finalize it after it is fully trimmed. You would not want any less than 45Deg combined unless it was heavy and over powered etc as often until the balance / motor thrust line is optimised, temporary trim can eat into elevator throw. Best to always optimise everything before using any other FC modes as problems can be masked.

DzikuVx, 2018-01-17:
Issue looks like solved, closing it. Thaks

closed 2018-01-17 by DzikuVx