iNavFlight/inav #2634: Motor output doesnt work

error414, 2017-12-30:
* SPracingf3 NEO, last version of INAV 1.8.

I flashed FW into FC and I only allowed PWM motor and servo output. Then I chose mixer "airplane". According to the picture, motors should be on output 1 and 2. When I look into the target file, motors should be on pins B08 and B09, which is the output ESC5 and ESC6. 

https://github.com/iNavFlight/inav/blob/master/src/main/target/SPRACINGF3NEO/target.c#L35

I connected the servo to the pin ESC5 for a test. The servo did not react to the motor slider. I found out that on pins ESC5 and ESC6 are mapped output for servo 2 and servo 4. 

pad ESC5 => servo 2
pad ESC6 => servo 4 

Motor isnt on any ESCx pad.

wx4cb, 2017-12-30:
IIRC, usually motors are on "motor1 and motor2" servos would be on 3/4/5/6 - have you checked the actual physical motor 1/2 pads to see, that should have the motor signals on it

digitalentity, 2017-12-31:
@error414 fixed-wing motors on ESC5/ESC6 is the only option that is compatible with the hardware without issues. For some boards pin mapping that is shown in the Configurator is wrong.

Documentation requires updating though. We might want a wiki section that will reflect pin mappings for supported boards.

closed 2017-12-31 by digitalentity