iNavFlight/inav #2636: OSD indication of loaded waypoints

Hwurzburg, 2017-12-30:
enhancement to add an indicator in the OSD that confirms a non-zero number of waypoints are loaded in volatile memory....ie to confirm that  valid waypoints were transferred from NV to RAM by stick command....anything will do... or even a message prior to arming after the cannot arm warnings are gone: "WAYPOINTS LOADED!" which disappears upon arming....either should be simple to implement....once # waypoints is implemented in the OSD mode display or enroute message in the future, then maybe that can be used also, ie " X Waypoints loaded!"

digitalentity, 2018-05-14:
@fiam @shellixyz @DzikuVx what do you think?

shellixyz, 2018-05-14:
It would be useful. I'll take a look.

fiam, 2018-05-14:
Would be easy to implement on top of a message bus

breadoven, 2021-11-29:
Closed by various PRs.

closed 2021-11-29 by breadoven