iNavFlight/inav #2638: No summary screen after disarming

compuatic, 2018-01-01:
Issue on inav 1.8.1
No summary screen on OSD after disarming following a long flight. Usually beyond 60km and 60mins+. Have not seen this with 1.7.2

Using omnibus F4 pro

mrcottonmouth, 2018-03-26:
I had the same problem 2 times today, same board, same inav version. No stats screen. I didn't seem to disarm either after reviewing the video. Maybe it's ok in 1.9?

shellixyz, 2018-03-26:
I can't find any bug that could cause this. The summary screen should be displayed unless the throttle or pitch sticks are pushed (high position). I will try to reproduce the issue.

fiam, 2018-03-26:
Maybe we should wait ~1 second to clear the stats if THR or PIT is already high when disarming.

shellixyz, 2018-03-26:
Yes that or wait for them to not be high before taking into account the high value to clear the stats.

shellixyz, 2018-03-27:
@compuatic @mrcottonmouth 
Please try this binary and tell us if it is working better for you:
[inav_1.9.0_OMNIBUSF4PRO.hex.zip](https://github.com/iNavFlight/inav/files/1850112/inav_1.9.0_OMNIBUSF4PRO.hex.zip)

mrcottonmouth, 2018-03-27:
I will try it at my 1st opportunity, which may be this weekend. I will go ahead and make the transition from 1.8 to 1.9 test rom and be ready for the weekend.

fiam, 2018-03-27:
@shellixyz I definitely prefer your idea. Better to check if THR/PIT are high on disarm and wait until they become low and then high again to clear the stats.

mrcottonmouth, 2018-03-31:
@shellixyz 
This did fix the issue. After a 71 minute flight, no issues. Summary screen was perfect. Thank you!