iNavFlight/inav #2642: Omnibus F4SD or F4 v3

gge0rgi0u, 2018-01-07:
Omnibus F4SD

INAV/OMNIBUSF4PRO 1.8.0 Nov  1 2017 / 06:34:32 (912d1315)

Flashing the board with Betafight Sbus receiver works fine on UART6

Flashing the board with Inav 1.8 omnibusF4v3 firmware, the usb stops working and two slow flashes followed be 3 quick flashes, appear on the blue led, repeatedly
Flashing the board with the alpha version for the F4SD gives the same problem as F4v3 firmware

Flashing the board with the Inav1.8 omnibusf4pro 1.8 the board works fine except no sbus is connected
on all ports

According to BF pin C07 is the Uart6 RX Sbus

resource   BEEPER 1 B04
--
resource MOTOR   1 B00
resource MOTOR   2 B01
resource MOTOR   3 A03
resource MOTOR   4 A02
resource MOTOR   5 A01
resource MOTOR   6 A08
resource PPM 1   B08
resource PWM 1   B08
resource PWM 2   B09
resource PWM 3   C06
resource PWM 4   C07
resource PWM 5   C08
resource PWM 6   C09
resource   LED_STRIP 1 B06
resource   SERIAL_TX 1 A09
resource   SERIAL_TX 3 B10
resource   SERIAL_TX 6 C06
resource   SERIAL_RX 1 A10
resource   SERIAL_RX 3 B11
resource   SERIAL_RX 6 C07
resource   INVERTER 3 C09
resource   INVERTER 6 C08
resource LED 1   B05
resource   SPI_SCK 1 A05
resource   SPI_SCK 2 B13
resource   SPI_SCK 3 C10
resource   SPI_MISO 1 A06
resource   SPI_MISO 2 B14
resource   SPI_MISO 3 C11
resource   SPI_MOSI 1 A07
resource   SPI_MOSI 2 B15
resource   SPI_MOSI 3 C12
resource   ESCSERIAL 1 B08
resource   ADC_BATT 1 C02
resource   ADC_RSSI 1 A00
resource   ADC_CURR 1 C01
resource   BARO_CS 1 B03

According to Inav pin C07 is free
IO:
----------------------
A00: FREE 
A01: MOTOR5 OUT
A02: MOTOR4 OUT
A03: MOTOR3 OUT
A04: MPU CS
A05: SPI1 SCK
A06: SPI1 MISO
A07: SPI1 MOSI
A08: MOTOR6 OUT
A09: SERIAL1 UART TX
A10: SERIAL1 UART RX
A11: USB IN
A12: USB OUT
A13: FREE 
A14: FREE 
A15: OSD CS
B00: MOTOR1 OUT
B01: MOTOR2 OUT
B02: FREE 
B03: BARO CS
B04: BEEPER OUT
B05: LED1 OUT
B06: FREE 
B07: SDCARD IN
B08: PPM IN
B09: FREE 
B10: I2C2 SCL
B11: I2C2 SDA
B12: SDCARD CS
B13: SPI2 SCK
B14: SPI2 MISO
B15: SPI2 MOSI
C00: INVERTER OUT
C01: FREE 
C02: ADC CH2
C03: FREE 
C04: MPU EXTI
C05: FREE 
C06: FREE 
C07: FREE 
C08: FREE 
C09: FREE 
C10: SPI3 SCK
C11: SPI3 MISO
C12: SPI3 MOSI
C13: FREE 
C14: FREE 
C15: FREE 
D00: FREE 
D01: FREE 
D02: FREE 
D03: FREE 
D04: FREE 
D05: FREE 
D06: FREE 
D07: FREE 
D08: FREE 
D09: FREE 
D10: FREE 
D11: FREE 
D12: FREE 
D13: FREE 
D14: FREE 
D15: FREE

wymbly1971, 2018-01-09:
I had the exact same problem with iNav 1.8.1. 

I was using a clone Omnibus F4 Nano board with a Frsky XM+ receiver. I checked 2 other known working Frsky rx's and same issue. The XM+ works perfectly fine with Betaflight 3.2.0 and up.

I "guessed" that the iNav firmware was not instructing the hardware invertor to work. I soldered a wire to the non-inverted side of the receiver and used that to connect to UART 6 and it works like a charm.

shellixyz, 2018-01-15:
I believe this has been fixed recently please try this: [inav_1.8.1_OMNIBUSF4V3.hex](https://drive.google.com/open?id=1M4oHqC0AWE8ZF_hzX2UWD8Sv2wyuhJfk)

wymbly1971, 2018-01-19:
That works!!! Thanks so much Shellixyz :)

...now what about the buggered up battery voltage display in the OSD? It shows a weird symbol where the numeric value should be.

shellixyz, 2018-01-19:
Great! The hex file I uploaded is not from a released version it has been built from the development branch so please be careful with it.

The oddities in the OSD display is more than likely because you need to update the OSD fonts to match the new OSD code.

wymbly1971, 2018-01-19:
That worked too :) Thanks so much

gge0rgi0u, 2018-01-19:
Thanl you for the help shellixyz
I havent used the file on my plane because I already uninverted sbus ans sport on the xsr.
I tried it on a spare fc and works fine.

closed 2018-04-25 by DzikuVx