iNavFlight/inav #2646: Matek F405 CTR (feature request)

raul-ortega, 2018-01-08:
It seems that "INAV1.8.0_MatekF405 listed in INAV configurator can’t support F405-CTR", as it says on the mateksys product page.

[http://www.mateksys.com/?portfolio=f405-ctr#tab-id-5](http://www.mateksys.com/?portfolio=f405-ctr#tab-id-5)

Mateksys has published its own hex file for this flight controller and I wonder if the source code has been added to the oficial development branch of iNav for next release...

[http://www.mateksys.com/Downloads/FW/inav_1.8.0_MATEKF405.hex](http://www.mateksys.com/Downloads/FW/inav_1.8.0_MATEKF405.hex)

Thanks

stronnag, 2018-01-08:
Please seehttps://github.com/iNavFlight/inav/pull/2566 (and if it meets your needs, please also close this issue).

raul-ortega, 2018-01-08:
Please keep it open, I would like to test the development version as soon as possible on my MATEK F04 CTR for an airplain and will close the issue if it really works.

mrduncan626, 2018-01-23:
I have a couple of the Matek F405-CTR on order for use with iNav.  Does the iNav firmware on the Matek website, as noted above, work on the this FC?

Sedgee, 2018-02-09:
I am looking at purchasing the same and would also be interested in knowing.

Azuxul, 2018-02-15:
Firmware form iNav configurator dosen't work but the firmware on the Matek web site work.
Tested with my Matek F405-CTR and inav_1.8.0_MATEKF405 firmware.

stronnag, 2018-02-15:
you need to wait for 1.9, or build the development branch.

digitalentity, 2018-02-16:
Closing, MATEK F405-CTR is supported by `development` and will be supported by 1.9

closed 2018-02-16 by digitalentity