iNavFlight/inav #2648: PASSTHROUGH mode improvements (expo/ servo rates)

giacomo892, 2018-01-08:
I'm thinking about something to enhance Fixed Wing experience when in PASSTHROUGH mode. 

Basically what pilots do, if they can, is to add some additional expo on the radio when they switch to PASSTHROUGH mode. This let them to control the aircraft in a much smoother way. 

I personally fly with FlySky radios and it does not let me adding additional expo.

Furthermore it is advisable not to use radio advanced functionalities when you use INAV. (like for instance trims and mixing) 

I came up with those solutions:

1. Add a new configuration parameter like `rc_passthrough_expo` that takes in account expo setting when in pass mode. It can be like 255 for disabled and then standard`rc_expo` will be used instead. 

2. Add a new configuration parameter like `passthrough_expo_boost`as a multiplier for the current `rc_expo` . Then pass expo will be  `passthrough_expo_boost` / 100 * `rc_expo` It can be 1 as default.

3. Append a new number to `servo` statement and specify there a new servo rate only used in PASSTHROUGH. This will also do the job without messing with expo but just limiting servo travel when in PASSTHROUGH mode. Idea from @fiam 

What do you think? Which idea do you like (if you do 💃 ) ?

gregorg, 2018-01-12:
Good idea :+1:

shellixyz, 2018-01-15:
I already made a patch that adds a `manual` mode with dedicated expo and rates. If you'd like to test it let me know. I will make a pull request for it soon.

giacomo892, 2018-01-16:
Great! Create a pull request so everyone can check :) Thanks!

shellixyz, 2018-01-17:
@junwoo091400 

I used the make command with an invalid target to display the list of targets:
```
Makefile:111: *** Target 'DONOTEXIST' is not valid, must be one of AIRBOTF4 AIRHEROF3 AIRHEROF3_QUAD ALIENFLIGHTF3 ALIENFLIGHTF4 ALIENFLIGHTNGF7 ANYFC ANYFCF7 ANYFCF7_EXTERNAL_BARO ANYFCM7 BEEROTORF4 BETAFLIGHTF3 BLUEJAYF4 CHEBUZZF3 CLRACINGF4AIR COLIBRI COLIBRI_RACE DYSF4PRO F4BY FALCORE FF_F35_LIGHTNING FISHDRONEF4 FRSKYF3 FRSKYF4 FURYF3 FURYF3_SPIFLASH KFC32F3_INAV KROOZX LUX_RACE MATEKF405 MATEKF405OSD MATEKF722 MOTOLAB OLIMEXINO OMNIBUS OMNIBUSF4 OMNIBUSF4PRO OMNIBUSF4PRO_LEDSTRIPM5 OMNIBUSF4V3 OMNIBUSF7 OMNIBUSF7V2 PIKOBLX PIXRACER QUANTON RCEXPLORERF3 REVO REVO_OPBL RMDO SPARKY SPARKY2 SPARKY2_OPBL SPRACINGF3 SPRACINGF3EVO SPRACINGF3EVO_1SS SPRACINGF3MINI SPRACINGF3NEO SPRACINGF4EVO STM32F3DISCOVERY YUPIF4. Have you prepared a valid target.mk?.  Stop.
```

Then I used a shell loop to build each one:
`$ for target in <target_list_copy_pasted>; do make TARGET=$target; done`

junwoo091400, 2018-01-17:
@shellixyz 
Thank you so much, I really appreciate it :)

Dronek, 2018-01-18:
> Basically what pilots do, if they can, is to add some additional expo on the radio when they switch to PASSTHROUGH mode. This let them to control the aircraft in a much smoother way.

Maybe I miss the point but why is that? RC EXPO is applied in PASSTHROUGH by default and a value of e.g. 0.7 works for PASSTHROUGH, ACRO and so forth. When I switch between ACRO and PASSTHROUGH the plane feels basically the same in terms of control. It is just more stable without stick input in ACRO mode but I don't see why more expo would be useful and all iNav pilots that I know in person don't apply additional expo in PASSTHROUGH.

In my opinion it would make more sense to adjust your mechanical linkage, servo travel and RC EXPO in such a way that flying in PASSTHROUGH feels comfortable. Then adjust the rates in ACRO until you get a desired behavior in this mode, no?

shellixyz, 2018-01-18:
Separate expo setting for me is not to increase it in passthrough (manual) but on the contrary less expo when using low rates (less than 100% control surface deflection).

Dronek, 2018-01-18:
> Separate expo setting for me is not to increase it in passthrough (manual) but on the contrary...

Understood, but why don't you set RC EXPO to work for you in PASSTHROUGH and then tune just the rates in ACRO to result in similar stick feel compared to PASSTHROUGH?

Redshifft, 2018-01-18:
Acro is a "rate" deg/sec control not a proportional control surface and simple to set to your required sensitivity. The other level modes are indirect through adjustable  pid controllers, Passthrough is the only untainted direct control that the plane feels you and you feel the plane.
As I said in the other thread I'm not even sure any "rate" settings are processed in passthrough any more - we used to have the RC rate scaler but I think that's locked at 1 now.
I imagine you guys have set up the planes to fly nice in Angle or Acro and then find Passthrough a disappointment where it should have been setup in Passthrough first which is the most important mode also the only one you can correctly trim your plane in.

shellixyz, 2018-01-18:
You have 3 switchable rate profiles for stabilized modes but for passthrough at the moment you are forced to use full control surface deflection. This feature request asks to fix this issue.

digitalentity, 2018-01-21:
Suggesting to move discussion about improved passthrough to the relevant PR #2661

closed 2018-01-21 by digitalentity