iNavFlight/inav #2649: Feature Request: Voltage/Throttle compensation

rodizio1, 2018-01-08:
With the Graphene Lipos I've been using, that annoying voltage sag (and thus having to compensate throttle position constantly with the battery getting more and more empty), is not a big problem anymore because the discharge curve of Graphene Lipos is quite flat.

However, now that Li-Ion cells with their not-so-flat discharge curve seem to be getting more popular, the problem re-appeared.

The idea is old: Compensate voltage sag inside the flightcontrol so that a given throttle position always results in the same rpm/thrust. I.e. if the voltage is 3.3v under load when the battery is near-empty and 4v under load when full, that would mean 3.3/4 = 0.825 or "when battery is at 4volts, apply target-throttle * 0.825".

My feeling is, this could probably not only give the pilot a more consistent throttle feel, but also help within the internals of the flightcontrol. For example, there is a hover-throttle value that has to be set to the amount of throttle the copter needs to hover for the alt-hold function. With Li-Ions, this is probably quite different with full battery compared to empty battery. If the flightcontrol would use the compensated throttle value internally, transition to alt-hold would probably work smoother.

The same applies for the cruise-throttle setting for planes, a cruise-throttle set just high enough to keep the plane in the air with full battery, could be too low with empty battery leading to a sagged tail or even worse stalling.

I think this would also give longer flight-times, as in the beginning with full battery, maximum throttle will be limited and thus less energy is used (Of course, you could also not use 100% throttle manually, but who does that ;))

Dronek, 2018-01-12:
I'd like to add some content based on my own experience with LiIon batteries and iNav on fixed wings.

Some facts:
- meanwhile many of the LiIon cells can be charged with a LiPo setting in the charger (4.2 V/cell)
- the minimum voltage under load is often specified as 2.5 V/cell
- for a four-cell LiIon battery this results in a voltage range from almost 16.8 V down to 10 V (you basically start a flight with a voltage equivalent to a full 4 cell LiPo and end the flight with a voltage equivalent to an empty 3 cell LiPo)
- LiIon cells are quite restrictive in terms of current draw and their voltage drops significantly when too much current is drawn

Therefore I support the idea of having a compensation for the wide voltage range. However, I suspect that multiplying the throttle value simply by a voltage ratio (as mentioned by @rodizio1) won't give satisfying results. 

Instead, it could make more sense to have a "constant power" mode where throttle is controlled by iNav in order to maintain a constant power value (power = current * voltage). In this mode the throttle stick could be assigned to control the power value in a user defined range (e.g. full throttle = 150 W).
This is only possible with a working current sensor but I guess that users of LiIon batteries are very likely to have this sensor anyway.

In navigation modes a fixed cruise throttle could indeed result in a stalling aircraft when LiIon batteries are used. A fixed cruise power would automatically compensate for a slowly dropping voltage by constantly increasing the throttle value.

Feldsalat, 2018-01-12:
I often use 2s Li-Ion cells in my models, planes and copters.
I only have problems with changing wind directions! (also with LiPo)

Voltage-dependent values ​​for "nav_fw_cruise_thr" and "nav_fw_min_thr" might be useful.
Especially "nav_fw_min_thr" is bit problematic in my planes.
I do not think that constant power would be helpful in changing wind directions!?

Dronek, 2018-01-12:
> I do not think that constant power would be helpful in changing wind directions!?

Keeping the electrical power constant is just another way of automatically increasing throttle with dropping voltage in order to keep the thrust at a more or less constant level. I just think it is a more versatile and controlled way than setting a constant voltage to throttle ratio.

@Feldsalat: As far as I understand your particular problem has more to do with varying groundspeed due to changing wind. It is not so much related to this issue here. Keeping a minimum power in navigation modes can prevent a plane from stalling, even if the voltage is low while a constant "nav_fw_min_thr" can result in a stall at low voltage (power too low) as well as in inefficient flight at higher voltage (power higher than necessary).

Feldsalat, 2018-01-12:
> @Feldsalat: As far as I understand your particular problem has more to do with varying groundspeed due to changing wind. It is not so much related to this issue here.

Was just one example of a real "throttle is controlled by iNav" problem.
Changing wind direction and throttle too low (=airspeed too low) causes the problem and that gets worse with decreasing battery voltage. So it is somehow related to this issue here... 
Waiting for: Implement wind estimation #2450 

When using "althold" I have almost no problems with decreasing battery voltage on planes.
But I have to set "nav_fw_min_thr" a bit higher. Voltage-dependent values ​​would be an easy way to help here. Because if value is set too high, i sometimes get unwanted climbing at fresh battery.

I normally have no current sensors in my Li-Ion powered planes.

shellixyz, 2018-01-16:
I tried to implement this but it is not simple. You would need a current sensor dedicated for the motors to measure the power consumed only by the motor(s) to regulate it. I personally put the current sensor in front of everything right after the battery to compute the overall consumed energy needed to estimate precisely the remaining battery energy.

The best in my opinion would be to leave the compensation to the ESC. The feature has been removed in BLHeli_S but with the old BLHeli you could enable closed loop control for the motor allowing an input signal to be matched to an RPM. This is what you want. Same input = same motor RPM = same thrust regardless of the battery voltage.

Dronek, 2018-01-16:
> The best in my opinion would be to leave the compensation to the ESC. The feature has been removed in BLHeli_S but with the old BLHeli you could enable closed loop control for the motor allowing an [...]

This is surely the most elegant way but at the same time it is quite unlikely to see this option coming soon to typical fixed wing ESCs since most of them don't even run BLHeli. 

> You would need a current sensor dedicated for the motors to measure the power consumed only by the motor(s) to regulate it.

After adding an RC filter to the signal line of my current sensor I get a very clean and steady current reading in my flying wing. Without any thrust it's around 0.8 A so it shouldn't be a problem to subtract this offset from the total current draw when flying. I don't see a big issue there as far as fixed wing aircraft is concerned.

shellixyz, 2018-01-16:
Maybe not a big problem I don't know but with a system current sensor you will be measuring the current to the servos and the motor RPM will vary with servos activity. You can't add a lot of filtering on the current measurement or the current and voltage measurements will be out of phase and will also cause unwanted variations in motor(s) RPM.

shellixyz, 2018-01-16:
BLHeli32 ESCs could make this relatively easy to implement in the FC with the ESC telemetry. I don't know enough about them but if the motor RPM can be red from the FC the closed loop RPM control should be easy enough to do.

digitalentity, 2018-01-21:
I'm not sure implementing this is trivial. If voltage sags under throttle further increasing throttle creates a feedback loop and makes things even worse. This needs to be carefully thought.

Dronek, 2018-01-22:
>  If voltage sags under throttle further increasing throttle creates a feedback loop and makes things even worse.

Good point. However, this can also happen when trying to keep airspeed constant or even constant ground speed, depending on battery state and condition. Thus it is a problem that could occur in different (future) auto-pilot situations and might need general safety measures (e.g. limiting the maximum power or current draw available for navigation modes).

Redshifft, 2018-01-22:
This is too complex to fully fix - The simple fix is to use the pretty flat voltage/discharge curve from our packs when used gently. Not what anyone wants to hear but get the Current down at the expense of peak performance obviously, and you will get much less variation with longer duration.
I have tested a lot of batteries in a rig and the claims are a joke.
Keeping away from warm motors/esc or batteries will keep you from the performance spiral.

Redshifft, 2018-01-22:
@Dronek  You are treating those LiIon cells well and flying for a long time, I doubt you could improve that much :)  I was thinking more regarding the Quad boys who are hitting the cells much harder.

closed 2018-05-27 by stale[bot]