iNavFlight/inav #2651: SP Racing F4 Evo, smart port telemetry not working.

flyandride, 2018-01-09:
I've flashed 3 boards, 2 real and 1 knock off, and have not been able to get smart port telemetry working.  Same boards, ports, receiver and wiring flashed with Betaflight and it works. Could I be doing something wrong?  
The F4 Evo seems like it would be a really good board for iNav as it's got 5 uarts.

davidngrc, 2018-01-09:
it seems I have the same problem.

flyandride, 2018-01-09:
I think I've got it figured out.  I read a post about another board with the sbus inversion not working.  The poster figured out that the inverter was not getting powered, so he powered it and had sbus working.  
I just checked power to the inverter with iNav and it's not powered.  Flashing it to Betaflight and the same pin shows 3.3V.  I will do some careful soldering and let you know if it works.

flyandride, 2018-01-09:
Didn't work.  I flashed it back to Betaflight and no telemetry until I cut the wire I so carefully soldered to VCC on inverter.  Maybe I could do an uninvert hack to the receiver.  Easy on X4RSB but very difficult on the X8R I'm using.

shellixyz, 2018-01-15:
If Smartport is working with Betaflight but not with iNav it could be a software issue. If you need the uninverted you can invert the inverted signal by building a discrete inverter. See https://quadmeup.com/smartport-inverter-for-f4-flight-controllers/

artusmuc, 2018-03-02:
Seems there a not enough INAV users for this FC to have a better support for this issue :/

closed 2018-05-27 by stale[bot]