iNavFlight/inav #2654: Possible Telemetry Issue using a DJI Style GPS

mrduncan626, 2018-01-12:
I have a Omnibus F4 Pro 3 (iNav 1.8.0 firmware) which I'm setting up with an SBUS Rx (FrSky XSR), S.Port Telemetry, and a DJI Naza GPS (clone).  In the iNav Configurator , I see all the GPS data but in the telemetry data stream on my Taranis (OpenTX 2.2.0) I see everything accept for the GPS data (Lat/long, GAlt, Number of Satellites, and GSpd.  Is this a known issue?

sergeysi779, 2018-01-13:
Have you discovered new sensors on Taranis after enabling GPS?

mrduncan626, 2018-01-13:
Yes, but only about 10 times :)

mrduncan626, 2018-01-14:
UPDATED Information. 

This is the data I see in the telemetry stream:
Acc XYZ
Tmp1
Tmp2 (the information in this looks to be valid - e.g. 2803, 2904, 3906 is a sample of some of the data)
A4
VFAS
Alt
Hdg 
VSpd
F000 (no idea what this is)
And, the usual A2, RSSI, and RxBt

I DO NOT see the following in the Telemetry stream:
GSpd
GPS
GAlt
0420

However, I do see valid GPS data (GSpd, GPS, GAlt) in the iNav Configurator GPS tab. And yes, I did do several re-Discover Sensors after installing the GPS/Compass.

mrduncan626, 2018-01-19:
UPDATE:  

Apparently, in order for the "Discover Sensor" to function completely on an Omnibus F4 Pro V3 and discover all the sensors, include all the GPS data (GSpd, GPS, GAlt, and 0420), you must enable the "Discover Sensor" function with the computer USB disconnected and only the Lipo connected.

closed 2018-01-19 by mrduncan626