iNavFlight/inav #2655: inav - TBS Discovery - GPS - Omnibus f4 v2

Christianmix, 2018-01-13:
Hello everyone, I searched for many places and I did not find anything about some mode of flight in the inav that keeps my TBS Discovery in a fixed position when I loose the stick, it is always drifting with the wind, is there any way that it stays like the naza or apm? When you release the stick, stop it and try to stay there until another command is given? I came to intstalar Betaflight and from what I could see seems to have the flight mode for gps, but does not have RTH in fail safe, does anyone know if there is a way to have gps flight in inav, or if there is a way to have RTH in betaflight?

Christianmix, 2018-01-13:
reading the manual of inav I was able to find out that I would only release the position hold modes if I had a magnetometer in my gps, then I got the gps of apm and installed and the flight mode appeared, the drone is stable, flies to the right, what can it be?

shellixyz, 2018-01-15:
Make sure you magnetometer is calibrated properly.

Christianmix, 2018-01-16:
it should not be, because with NAZA v2 my gps is pointing to 55 minutes, and with this this pointing to 42 minutes I managed to fly still following a tip of an inav user on youtube, I tried to calibrate several times and always continues same, but well, after pointing to the correct side and he stops walking alone, I wanted to know how to regulate so that he is more caught, as in my naza, I come at a speed of 40km / h and I loose the radio stick and the drone stops immediately and even with wind it remains in place, even if it pushes it back, and with the inva it is not so fixed so, after releasing the erle stick still walks about 2 meters and stops, and after that it is drifting a little according to the direction of the view, where can I improve it?

Christianmix, 2018-01-16:
How do I see if the bus is pointing north along with the aircraft?

https://youtu.be/HdEN7NCssmI

shellixyz, 2018-01-16:
I think you can check the north in the sensors tab but I'm not sure about the values to look for.

To improve `POSHOLD` handling you can tweak these values: `nav_mc_pos_xy_p`, `nav_mc_pos_xy_i` and `nav_mc_pos_xy_d`. Increasing `nav_mc_pos_xy_p` and `nav_mc_pos_xy_d` should increase the aggressiveness of the GPS control.

digitalentity, 2018-01-21:
@Christianmix do you have any results calibrating your compass?

Christianmix, 2018-01-29:
caliber and made other modifications, improved but it was not perfect, I left this project for now, I will do it another time with great patience and when I have a new gps, because in that I had to do a pull up, maybe the magnetometer of this old GPS is not so coming.

closed 2018-01-29 by Christianmix