iNavFlight/inav #2656: Omnibus f7 pro

Sallam84, 2018-01-14:
I flashed inav 1.8 on (omnibus f7 pro)  using omnibus f7 as the target, i had proplems with the ppm pin, after few hours troubleshooting i discovered the resources A03 for ppm is the issue , while it should be used on C05. Is there any way i can remap the resources so that C05 is ppm.

shellixyz, 2018-01-15:
You can't do it by software but you can build a custom version with the PPM pin edited to match your needs in `target/OMNIBUSF7/target.c`. If the OMNIBUSF7 FC really uses the C5 pin as PPM input the change should be pulled in the project source code. Can your board be a new variant of the OMNIBUS F7 ?

Sallam84, 2018-01-16:
Its a (clone)  from banggood (omnibus f7 pro).

closed 2018-01-16 by Sallam84

peter-umb, 2018-05-13:
I also have the banggood omnibus f7 pro and I am trying to setup ppm. Where do you see the pinout mapped to C05? Did you make a custom build or pull request using this pin?

shellixyz, 2018-05-13:
I guess he checked with a multimeter. No changes have been made to the PPM input pin since this issue has been opened. The pin used for the PPM input is PA3 which should correspond to the UART2 RX line according to the comments in the target files.

peter-umb, 2018-05-13:
My mistake. It looks like UART1 is already mapped to PA10. There must be some other issue happening.
     #define UART1_RX_PIN PA10

shellixyz, 2018-05-13:
Doesn't look like the right pin indeed

shellixyz, 2018-05-13:
Two pins should be connected to the RC input pad. One for SBus (probably the one you found) and another one for PPM.

shellixyz, 2018-05-13:
My guess is that it is PE13 since it was set as PE13 before but it has been changed to UART2 RX to solve a timer conflict with PWM output 3

peter-umb, 2018-05-13:
I just double checked and I can only find 1 pin connected to PPM. Although, I may have mistakenly identified PA10 when it was actually PA9. I have to run out and get the wife dinner but maybe I will try compiling with PA9 to see what happens when I get back?

peter-umb, 2018-05-13:
Just saw your comment. What version was it set to PE13? Maybe I can just try flashing that instead.

peter-umb, 2018-05-13:
Lol. I went a little crazy and tried the oldest/middle/newest from Cleanflight, iNavFlight, and Betaflight. None worked.

peter-umb, 2018-05-13:
[image]

Well, I checked a third time it looks like PA9 is the only pin that beeps with my continuity detector. It would seem that this device will just not work with available firmware and PPM. I will contact HobbyKing and see if they have any ideas. (It was HobbyKing and not BangGood. My bad - I am helping a friend put this together and he just shared his order details with me)

shellixyz, 2018-05-13:
It is just a poorly designed board. I guess it just have not been designed with PPM in mind. If you connect your PPM receiver to the UART2 RX line (the pin configured as PPM input) it should work.

peter-umb, 2018-05-14:
Forgive my naivete, but would you happen to know where the UART2 is on this board? The pinouts for this board are horrible. Maybe I can just use my trusty continuity trick to map the pin back to a port.

peter-umb, 2018-05-14:
rx3 - pa13
rx6 - pb3

UART2_RX_PIN - PA3  # NO CONTINUITY
UART3_RX_PIN - PB11  # NO CONTINUITY
UART4_RX_PIN - PA1  # NO CONTINUITY
UART5_RX_PIN - PD2  # NO CONTINUITY
UART6_RX_PIN - PC7  # 3rd pin from left AKA PWM 4

[image]

MorrisR2, 2022-02-16:
> Correct me if I am wrong, but the solution is to either buy a TX that supports IBUS/SBUS OR ditch this Omnibus f7 in favor of something that actually supports PPM

Between those two, lean towards getting away from PPM.

> I assume f4's are nice?

The original Airbot Omnibus F4 boards ($30 direct from the manufacturer) are well supported.