iNavFlight/inav #2657: Weird digital servo behavior

pmarkiewicz, 2018-01-15:
OMNIBUS F4 PRO LEDSTRIP M5, latest firmware. Flying wing preloaded with wing wing. No custom mixers.
Moving elev moves servo correctly but sometimes servo moves to end and return slowly.
Servor powered by external BEC, it's not problem with ground (same behavior when gnd from FC instead of BEC). No problem with power form BEC (checked voltage)
Not replicable with analog servos. It's not servo problem I checked 6 different DS Corona servos.

Video: https://www.youtube.com/watch?v=Th85OD2IZS8

shellixyz, 2018-01-15:
Would be great if you could check the output of your BEC and the signal wire with an oscilloscope. I don't think it can be software related. Are you sure you power rail doesn't sometimes dip when your BEC is loaded ? It happened to me with an OVC protected BEC. Sometimes the starting current was tripping the current protection and the tension collapsed making the servo twitching.

pmarkiewicz, 2018-01-16:
I don't have oscilloscope, but I can try to power servos from FC.

shellixyz, 2018-01-16:
Could be a good test if your FC can provide enough current. I don't recommend doing that as a definitive fix though. Alternatively if you can try with a bench power supply or a mains powered brick DC-DC it could be a good option.

pmarkiewicz, 2018-01-17:
Case closed. I replaced BEC and servos works.
Maybe this should be included in Wiki as I never expected that level of issues because to BEC.

shellixyz, 2018-01-17:
Great! Noted. I will add something about this issue in the wiki. If we are done with this issue please close it.

pmarkiewicz, 2018-01-17:
solved

closed 2018-01-17 by pmarkiewicz