iNavFlight/inav #2658: Omnibus F4 V3 - Can't enable softserial

DangerD256, 2018-01-15:
Softserial is enabled in configuration tab (also checked in cli) but port doesn't appears in ports tab

shellixyz, 2018-01-15:
That's because no software serial pins are configured for the OMNIBUS F4 V3 FC.

DangerD256, 2018-01-15:
Any way to change this? If I flash pro firmware they will work...

shellixyz, 2018-01-15:
No it probably won't work because on the V3 board one of the pins used for softserial on the PRO is used for switching the UART6 inverter. Free usable pins would need to be found to assign to the softserial port for the V3.

DangerD256, 2018-01-15:
PWM5 and 6. If I will flash firmware from omnibus f4 pro to my v3 it works.

shellixyz, 2018-01-16:
Odd. I don't understand how it could as PC8/PC9 used for softserial on the PRO doesn't seem to be connected to the PWM 5 and 6 outputs. Referring to the source code they should be connected to the CH5/CH6 pads on the PRO (CH5/CH6 is not PWM5/6, not the same thing).

shellixyz, 2018-01-16:
Also assuming you were talking about CH5/6 if you are using SBUS and the PC8 pin is really used for controlling the UART6 inverter it would cause issues with your RX link.

KnuckleUpFPV, 2018-01-19:
My uart6 doesnt work at all. no matter what I do. no rc link, no gps, no smart audio, nothing

shellixyz, 2018-01-19:
@KnuckleUpFPV Off topic. This issue is about softserial. Maybe something happened to your UART6 and it is not dead. Can't think of anything else. I know it is working for at least SBUS for other people.

KnuckleUpFPV, 2018-01-19:
Soft serial isnt working either. Meant to include that. And no the wires arent swapped. Like I said it works on uart 1 but uart6 will not respond to any input. Sbus, crsf, non inverted sbus. Nothing. Zip zilch nodda. My apologies for being off topic. No one knows why this uart doesnt work. I've been posting anywhere I can get a response. A lot of threads are dead. I'll just throw it out and buy something else. I have 18 other mini quads and 3 wings to fly.

KnuckleUpFPV, 2018-01-19:
Disregard. They removed ch5 and ch6. I scraped the board from the two very tiny pads by the sd card. Where the ch5 and ch6 were at on the v2. I soldered to them and gps is now working over soft serial. I hope someone else comes here and finds this

ekitru, 2018-01-21:
The problem looks like similar to mine.
I have Matek 405-AIO  board and trying to connect frsky S.BUS  telemetry throught softserial.
Enabling softserial feature does not help, there is not softserial ports at all.

KnuckleUpFPV, 2018-01-21:
I got soft serial working on the matek. Had to use cli to turn it on. The slider doesnt work in inav

ekitru, 2018-01-21:
I have tried feature -SOFTSERIAL, save, feature SOFTSERIAL, save
it does not help.
I am newbie in RC and INAV
Can you help me, may be some other feature need to enabled or disabled, or some command is needed to activate s.bus on pin6 ?

KnuckleUpFPV, 2018-01-21:
First can I ask why you aren't using the non inverted signal from the receiver? Its alot easier. You can pull the none inverted and hook it to a regular uart. And I use the cli command and moved the soft serial slider both. I had to move slider save and then do the cli command. It would be alot better to pull the non inverted signal. There are lots of pics on Google where to solder too. I'm happy to help you. I can't figure out how to post pics in here. You can find me on Facebook and we can use Facebook messenger. My name is Robert Criteser.  There is only two of us on Facebook. Myself and my uncle. The pic is of myself and my girlfriend at a new years party. Message me and I can send you pics.

shellixyz, 2018-01-21:
@ekitru S.Bus on UART6 doesn't work in 1.8.0. This is a known bug (see #2642 for an explanation an alpha fix). Also softserial doesn't work with the Omnibus F4 V3 board as no pins has been defined for the RX/TX lines.

KnuckleUpFPV, 2018-01-21:
He is talking about the matek f405 aio. Not the omnibus

KnuckleUpFPV, 2018-01-21:
If you use the none v3 firmware you can soft serial on the v3. The pads to solder to, are next to the sd card. They are very small and have to be scraped clean first. They are the spots where ch5 and ch6 is on the v5 version of the board. I have mine working.

ekitru, 2018-01-21:
I found a non-inverter hack for frsky receivers, but I am using R9 receiver. 
propably there is  a inverter chip c04f on second level board, but currently there is not instructions to how get non-inverted signal from r9 module.

ekitru, 2018-01-21:
I also took r9 because of some reviews which shows really good range and stability. 
Currently I can't find anythinf about uninverted hak for this module. It is took new =(

yohihoy, 2018-01-22:
Ok I have external inverter for frysky x8r it works in betaflight but it isnot working here. 
My hardware inverter from here, and again telemetry works connected to pwm5 pwm6:
https://github.com/iNavFlight/inav/blob/master/docs/Telemetry.md
But this command not working here as described:
set smartport_uart_unidir = ON
set telemetry_inversion = OFF
Where I wrong or could you please consider to make different firmware for extending our FC to use pwm5/6 mapping. Cause here only 3 UARTS. I cannot use my telemetry with INAV.

yohihoy, 2018-01-22:
As was expected, when sbus inv set to off, this disabled inversion on uart6 and receiver stop working.

KnuckleUpFPV, 2018-01-22:
So you don't have a hardware inverter? They can be made very easy. Run both with inverters and sbus inversion off.

yohihoy, 2018-01-22:
> So you don't have a hardware inverter? They can be made very easy. Run both with inverters and sbus inversion off.

Have you read my comments above where I mentioned this twice?
> Ok I have external inverter for my frysky x8r

> My hardware inverter

KnuckleUpFPV, 2018-01-22:
First off if you are going to be a dick you can find help elsewhere. I meant a hardware inverter built into the board. It's not a hard concept. If using an x8r you could just use the uninverted signal. Good luck to you.

KnuckleUpFPV, 2018-01-22:
And soft serial works fine on the v4. Had you read above you would see to use the pro target and the small pads by the sd card. It takes a cli command to activate it. And some tiny soldering. Pwm5 and pwm6 are not the pads you soft serial too. When you decide you want to see pics of how its done you can find me on Facebook as I stated above and I can send a pic. Otherwise maybe one of the other gentlemen will help you.

yohihoy, 2018-01-22:
@KnuckleUpFPV 
Buddy, I'm just trying to keep patience, reading any advices from any people, trying to apply them even they looks strange cause Im very interested, but whats wrong with you? I just point you so I mentioned that I have hw invertion for s.port telemetry, but it doesnt work just it. About pads, they are too small for me dont think I can keep FC unbroken when touch them :smile: 
But anyway thank you for your help.

KnuckleUpFPV, 2018-01-22:
Again you missed what I was talking about. A physical hardware inverter on the fc. Inverting the inverter makes it none inverted. Can't make this shit up. Using it none inverted through cli and two external inverters will net two none inverted signals. Do some research on sbus and how it works and you may understand.

yohihoy, 2018-01-22:
@KnuckleUpFPV 
Are we talking about the same thing, just to be sure? On my FC omnibusf4v3 s.port telemetry, workin only with external hw inverter like in article I linked. For sbus connected to ppm(uart6) FC already has hw inverter, but Im trying to fix s.port telem and map it to sft serial.

KnuckleUpFPV, 2018-01-22:
🖒

yohihoy, 2018-01-23:
@shellixyz 
> Free usable pins would need to be found to assign to the softserial port for the V3.

> Also assuming you were talking about CH5/6 if you are using SBUS and the PC8 pin is really used for controlling the UART6 inverter it would cause issues with your RX link.

I can confirm if it helps someone to make this FW better, softserial with discrete inverter for s.port X8R works in BF with PW5/PWM6 with this cli commands:
resource motor 5 none
resource motor 6 none
resource led_strip none
resource serial_tx 11 a1
resource serial_rx 11 a8

KnuckleUpFPV, 2018-01-23:
You would have to build a custom firmware. No mapping in inav.

shellixyz, 2018-01-23:
@yohihoy You don't need an inverter for S.Port on softserial

dolphin78, 2018-02-01:
If anybody's interested here's a custom firmware for genuine Omnibus F4 V3 with software I2C enabled on PWM5(SCL) and PWM6(SDA). This frees up UART3 as you can use PWM5/PWM6 instead of SCL/TX3 and SDA/RX3 for I2C. This firmware is based on OMNIBUSF4PRO instead of OMNIBUSF4V3 as I had major issues with compass orientation (BN-880) with original OMNIBUSF4V3 firmware.

Download link: [inav_1.8.0_OMNIBUSF4V3_I2C_REMAP.hex](https://raw.githubusercontent.com/dolphin78/inav/feature/OMNIBUSF4V3_I2C_REMAP/obj/inav_1.8.0_OMNIBUSF4V3_I2C_REMAP.hex)

Source code: https://github.com/dolphin78/inav/tree/feature/OMNIBUSF4V3_I2C_REMAP

KnuckleUpFPV, 2018-02-01:
Dolphin you are awesome. I'm currently soldered to the micro pads on bottom that I scraped clean. I will likely pull them and move them over with your build. Thank you for this. You are an absolute help to the community offering a solution like this.

yohihoy, 2018-02-02:
@dolphin78 
Do we need to enter some cli command, so this not working out of the box?

dolphin78, 2018-02-02:
no CLI command is needed. Did your compass work with this board on RX3/TX3 pins before you flashed this firmware?

yohihoy, 2018-02-02:
Yes, it worked

dolphin78, 2018-02-02:
I'll compare your settings with mine when I get home today.

dolphin78, 2018-02-03:
@yohihoy, I don't see any issues with your config, below is mine for comparison. 

I can think of only two things:
1. try to swap SCL/SDA wires
2. you may need pull up resistors on SDL/SDA lines

What is the compass you are using?

<details>
<summary>
My working settings and resources
</summary>

`# diff`

`# version`
`# INAV/OMNIBUSF4V3_I2C_REMAP 1.8.0 Feb  1 2018 / 11:56:07 (912d131)`

`# resources`

`# mixer`

`# servo`

`# servo mix`

`# feature`
`feature GPS`
`feature PWM_OUTPUT_ENABLE`

`# beeper`

`# map`

`# name`
`name Vlad's Quad`

`# serial`
`serial 0 2 115200 38400 0 115200`
`serial 2 1 19200 38400 0 115200`
`serial 5 64 115200 38400 0 115200`

`# led`

`# color`

`# mode_color`

`# aux`
`aux 0 2 1 900 1450`
`aux 1 3 2 1275 1675`
`aux 2 9 2 1275 1675`
`aux 3 8 2 1700 2100`
`aux 4 19 3 1575 2100`
`aux 5 23 0 1650 2100`

`# adjrange`

`# rxrange`

`# master`
`set gyro_sync = ON`
`set gyro_hardware_lpf = 188HZ`
`set gyro_lpf_hz = 90`
`set gyro_notch1_hz = 200`
`set gyro_notch1_cutoff = 100`
`set acc_hardware = MPU6000`
`set acczero_x = 117`
`set acczero_y = 8`
`set acczero_z = -77`
`set accgain_x = 4115`
`set accgain_y = 4076`
`set accgain_z = 4011`
`set align_mag = CW270FLIP`
`set mag_hardware = HMC5883`
`set magzero_x = 7`
`set magzero_y = -181`
`set magzero_z = 25`
`set baro_hardware = BMP280`
`set pitot_hardware = NONE`
`set receiver_type = SERIAL`
`set serialrx_provider = IBUS`
`set min_throttle = 1025`
`set max_throttle = 1900`
`set motor_pwm_rate = 8000`
`set motor_pwm_protocol = MULTISHOT`
`set failsafe_procedure = RTH`
`set gps_provider = UBLOX7`
`set gps_sbas_mode = AUTO`
`set nav_auto_speed = 833`
`set nav_manual_speed = 833`
`set nav_landing_speed = 500`
`set nav_rth_alt_mode = MAX`
`set nav_rth_altitude = 2000`
`set osd_rssi_pos = 23`
`set osd_throttle_pos = 65`
`set osd_gps_speed_pos = 2104`
`set osd_home_dir_pos = 2414`
`set osd_altitude_pos = 2276`
`set osd_vario_num_pos = 2295`
`set osd_ontime_flytime_pos = 2048`
`set osd_rtc_time_pos = 2072`
`set tz_offset = -300`

`# profile`
`profile 1`

`set mc_p_pitch = 58`
`set mc_i_pitch = 50`
`set mc_d_pitch = 35`
`set mc_p_roll = 43`
`set mc_i_roll = 40`
`set mc_d_roll = 30`
`set mc_p_yaw = 70`
`set dterm_lpf_hz = 80`
`set dterm_notch_hz = 200`
`set dterm_notch_cutoff = 100`
`set roll_rate = 67`
`set pitch_rate = 67`
`set yaw_rate = 67`

`# resource`
`IO:`
`----------------------`
`A00: FREE `
`A01: FREE `
`A02: MOTOR4 OUT`
`A03: MOTOR3 OUT`
`A04: MPU CS`
`A05: SPI1 SCK`
`A06: SPI1 MISO`
`A07: SPI1 MOSI`
`A08: FREE `
`A09: SERIAL1 UART TX`
`A10: SERIAL1 UART RX`
`A11: USB IN`
`A12: USB OUT`
`A13: FREE `
`A14: FREE `
`A15: OSD CS`
`B00: MOTOR1 OUT`
`B01: MOTOR2 OUT`
`B02: FREE `
`B03: BARO CS`
`B04: BEEPER OUT`
`B05: LED1 OUT`
`B06: FREE `
`B07: SDCARD IN`
`B08: FREE `
`B09: FREE `
`B10: SERIAL3 UART TX`
`B11: SERIAL3 UART RX`
`B12: SDCARD CS`
`B13: SPI2 SCK`
`B14: SPI2 MISO`
`B15: SPI2 MOSI`
`C00: FREE `
`C01: FREE `
`C02: ADC CH2`
`C03: FREE `
`C04: MPU EXTI`
`C05: FREE `
`C06: FREE `
`C07: SERIAL6 UART RX`
`C08: INVERTER OUT`
`C09: FREE `
`C10: SPI3 SCK`
`C11: SPI3 MISO`
`C12: SPI3 MOSI`
`C13: FREE `
`C14: FREE `
`C15: FREE `
`D00: FREE `
`D01: FREE `
`D02: FREE `
`D03: FREE `
`D04: FREE `
`D05: FREE `
`D06: FREE `
`D07: FREE `
`D08: FREE `
`D09: FREE `
`D10: FREE `
`D11: FREE `
`D12: FREE `
`D13: FREE `
`D14: FREE `
`D15: FREE `
</details>

yohihoy, 2018-02-03:
1. already tried
2. need to check with resistors

Im using HMC5983 from Honeywell as part of gps "radiolink se100".

jdd1777, 2018-02-05:
Thank you dolphin78,Your hex worked perfect. Now i have everything I want working.

jasoneyes01, 2018-02-24:
Dolphin, I'm glad I found this! Setting up an S800 with V3 board and 880 GPS. Can I use UART3 TX for the Smart Audio on my VTX with this custom firmware?

dolphin78, 2018-02-26:
jasoneyes01, never tried it myself. You can try it and share results with others.

jasoneyes01, 2018-02-26:
The new V3 should be here this week, will post the results.  Had alot of problems with the V2 PRO getting everything working not only to have the micro USB fall off the board. Plan B is the V3 board with 2 adjustable DC to DC Buck regulators for 6 volts and 10 volts to power servos and video.

jdd1777, 2018-02-28:
So dolphin, any plans on a 1.9. Thanks

evmoius, 2018-03-01:
FYI I'm having the same issue with my compass as @yohihoy with your custom hex @dolphin78 .  It's an improvement, as before trying it I couldn't have ANYTHING plugged into a UART other than my sbus.  Trying to turn on  anything else like GPS would cause my receiver to just quit. 

Wondering if either of you ever figured out what was happening?  Thanks for taking the time to make a custom solution though!

SebiCo, 2018-03-02:
Big thanks to @dolphin78 for source code.
[inav_1.9.0_OMNIBUSF4V3_I2C_REMAP.hex](https://www.sendspace.com/file/ej7gvg)

romko1, 2018-04-04:
To dolphin78: Hello,
I can not create my own firmware for FC OmnibusF4v3.
Can you help me?
I need to change the following:
A01: MOTOR5 OUT - change to softserial 11 TX, A08: MOTOR6 OUT - change to softserial 11 RX,
B05: LED1 OUT - change to softserial 12 TX, B06: FREE - change to softserial 12 RX.
Will you create my own hex file?
Thanks.

rapiertg, 2018-04-23:
Hi,
Is there any chance to get i2c remapped for 1.9.1 or even make it an official target?

jdd1777, 2018-04-23:
I would also appreciate this made as an official target also at least until we get resource remapping. Thanks for all the development you guys have been doing for us. I love inav and see it growing in the community.

cyrus2047, 2018-04-26:
Hi,
I try to remap PWM5-> RX and RWM6 -> TX basic on 1.9.1. 
I just have a test in front of my PC only, my gps can work with softseria1. Would someone help to have a detail test?
Thanks a lot.
[inav_1.9.1_OMNIBUSF4V3(PWM5[Rx]_PWM6[Tx].zip](https://github.com/iNavFlight/inav/files/1949328/inav_1.9.1_OMNIBUSF4V3.PWM5.Rx._PWM6.Tx.zip)

Bernard-S, 2018-04-27:
Hi
I'm trying to connect Bluetooth to an Omnibus F4 V5, do I need to use softserial or can I use one of the spare Uarts?  On my SPRacing F3 board I just plugged into uart2 set the baudrate and it worked.  No such luck with the Omnibus.

shellixyz, 2018-04-27:
@Bernard-S You should only have to connect the BT module to the UART, enable MSP on the UART you connected it to and set the baud rate to the same baud rate you configured your BT module.

It not recommended to use softserial to connect a bluetooth module. I think the max baud rate for SS is 19200. Might work with higher speed on F4 boards but I'm not sure.

Bernard-S, 2018-04-28:
Thanks for your help but the problem I have is that now I don't seem to have any spare UART connections left for the HC05.  I thought I could use UART6 (RX6/TX6) but have now read somewhere that these are used for the SBUS receiver, is that right ?

UART1 is used for GPS and UART3 is used for the compass.  Are there any other connections I could use ?

I'm running INAV 1.9.3 configurator and inav_1.9.1_OMNIBUSF4PRO_LEDSTRIPM5.hex

rapiertg, 2018-04-29:
Built 1.9.1 with @dolphin78 changes to remap i2c to pwn5 and pwm6 but after enabling magnetometer all I get is red icon and constant beeping. This was working with 1.8.9 and 1.9.0 supplied here. Were there any changes in code that needs adressing beside @dolphin78's one or am I doing something wrong? Have someone successfully build remapped i2c 1.9.1?

shellixyz, 2018-04-29:
@Bernard-S Yes UART6 is used for SBus. If you have a GPS on UART1 and a compass on I2C (shared with UART3) then you don't have any more free UART. Your only option would be to use a soft serial port but these are limited and also not available by default on OMNIBUSF4V3 (the target should should use with an Omnibus F4 V5, not OMNIBUSF4PRO). If you need more hardware UARTs you should use another FC.

Bernard-S, 2018-04-30:
What about the possibility of reassigning PWM5&6 ?  Most people are not likely to use these outputs whereas compass, GPS and bluetooth must be quite common ?

Living in hope :-)

shellixyz, 2018-04-30:
@Bernard-S It is possible but then I think your only viable option would be to use GPS on soft serial with reduced baud rate. Personally I would not do that.

krzysztofmatula, 2018-05-25:
This patch (https://github.com/iNavFlight/inav/pull/3222) aims to add support for unidirectional (TX-only) softserial on OMNIBUSF4V3 by using UART6-TX pin. Once enabled, UART6-RX can still work as a HW UART e.g. for S.BUS, while UART6-TX can be configured for any other purpose, e.g. for one-way telemetry.

KnuckleUpFPV, 2018-05-25:
So you are using a uart for soft serial? It's easier to use the pro firmware and the two small pads by the sd card. The stock tip on the ts100 was enough to solder to them. Some hot glue helped support them. Did a write up on the inav facebook page with pics to the two pads I'm talking about.

[84 more replies]

closed 2019-02-04 by DzikuVx