iNavFlight/inav #2659: omnibus f7 ppm

eeprommemory, 2018-01-16:
omnibus f7
INAV/OMNIBUSF7 1.8.0 Nov  1 2017 / 06:59:03 (912d1315)

ppm (pe13) does nothing.
also testing neo m8n + mag akm8975 on RX3 TX3 (sdl sca) nothing gps half on RX6 TX6 works.
have photo of insides it has a m8n + a akm8975 and some cpu with the writing ground off

eeprommemory, 2018-01-19:
gps with mag problem fixed see post #2619.
PPM on pe13 still not working :(

DzikuVx, 2018-01-19:
@eeprommemory PPM is wired to PA3 not PE13. UART2 has to be disabled for PPM to work

eeprommemory, 2018-01-19:
i used multi-meter to trace ppm pin to pe13

eeprommemory, 2018-01-19:
also tried it on rx1 pa10 and that did not work.

DzikuVx, 2018-01-20:
This is board design. Ppm is wired to both pe13 and pa3. Inav uses pa3.
At least that was in v1 . Notsure about clones

eeprommemory, 2018-01-21:
pa3 is not connected.

eeprommemory, 2018-01-21:
ebay seller sold it as a omnibus f7 with current sensor.
this looks like the same thing.
https://www.banggood.com/Omnibus-F7-Pro-Flight-Controller-Built-in-Dual-Gyro-AIO-OSD-Current-Sensor-and-LC-Power-Filter-p-1169518.html?cur_warehouse=CN
[image]

DzikuVx, 2018-01-21:
OK, so this is chinesium clone and looks like we are not supporting it fully. Original Airbot Omnibus F7 works with PPM.

eeprommemory, 2018-01-21:
came loaded with betaflight.
everything works but ppm and select  ICM20608.
gps was the same deal ebay seller sold it as a neo m8n + mag no instructions i had to open it to get the pinouts and spotted the chip with the writing ground off when the mag did not work i knew the chip had to go.

Sallam84, 2018-01-21:
Just solder the ppm wire to A03 leg and you are good to go.

shellixyz, 2018-01-21:
@eeprommemory You can try this. It is a custom build with PE13 as PPM input: [inav_1.8.0_OMNIBUSF7.hex](https://drive.google.com/open?id=1hZrF9nkG_3-xQdjU0CdaiPtTdUx9nHws)

eeprommemory, 2018-01-22:
did not work set rx to ppm and set one port at a time to rx.
UART 1 not used
UART 3 mag ak8975  working
UART 6 ublox m8n working
baro working
mpu6000 working
 ICM20608 don't know set gyro_to_use = 1 did not work
4 motors working

# version
# INAV/OMNIBUSF7 1.8.0 Jan 21 2018 / 16:54:59 (912d1315)

# resources

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_UIB
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature GPS
feature RSSI_ADC
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
[... 378 lines cut ...]
set nav_mc_pos_z_d = 0
set nav_mc_vel_z_p = 100
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

eeprommemory, 2018-01-22:
grabbed my old cc3d and loaded inav_1.7.3_CC3D_PPM1.hex and it worked.

eeprommemory, 2018-01-22:
loaded old firmware inav_1.8.0_OMNIBUSF7 and bodged a wire to PA3 and it worked this is not a real good solution because someone else may have same problem.
adding a wire from PA3 to PE13 wont be easy so i never tried it.

eeprommemory, 2018-05-10:
hate it when that happens.
3.3v reg will be toast.

closed 2018-07-23 by stale[bot]