iNavFlight/inav #2660: inav 1.6 msp communication

amitmanish, 2018-01-16:
I am trying to implement serial communication between naze32 rev and raspberry pi. Naze32 has inav 1.6 firmware. My ultimate goal is to control my drone i.e naze32 using raspberry pi. I tried pymultiwii library with my receiver type rx msp in inav configurator. What am i doing wrong

shellixyz, 2018-01-16:
Never done this but check if you are connected to the right serial port (the one you enabled MSP on) and see if you can read data with MSP to check if your connection is right.

amitmanish, 2018-01-17:
I can read data like accelerometer value but cant arm drone using raspberry pi. Is msp communication one way.

shellixyz, 2018-01-17:
No it is not. MSP is the same protocol used by the configurator. You should read the MSP source code on the iNav side to make sure the commands are right/up-to-date on the Pi side.

amitmanish, 2018-01-17:
Thanks shellixyz. I wiill look into the msp source code of inav.

closed 2018-01-18 by amitmanish