iNavFlight/inav #2663: Clear documentation about MavLink integration

glatorg, 2018-01-18:
I've been messing around for a month trying to get it work but without success :-(
I would like to add SCALED_IMU values (xacc, yacc, zacc, xgyro, ygyro, zgyro) to the MavLink messages sent on the ground.
I understand that I have to build my own iNav image with Cygwin
I understand that I have to modify main/telemetry/mavlink.c file
But can't understand how the file works, I asked to RcGroups forum and Facebook iNav group without answer. They suggest me to post here.
I have written something like but without success:

void mavlinkSendImu(void) { mavlink_msg_imu_pack(mavSystemId, mavComponentId, &mavSendMsg, millis(), SCALED_IMU(xacc), SCALED_IMU(yacc), SCALED_IMU(zacc), SCALED_IMU(xgyro), SCALED_IMU(ygyro), SCALED_IMU(zgyro); mavlinkSendMessage(); }

Please can someone explain me how to complete my need? Or maybe someone can build the image for me? I use Naze32 rev6.
If someone accepts donations, come forward!

stronnag, 2018-01-18:
Did you not  read the followup post on RGC?
Did you try its suggestions?

https://www.rcgroups.com/forums/showpost.php?p=38979128&postcount=17436

glatorg, 2018-01-18:
Sure, I read it but the topic (justly) is going fast in many other questions.
I would like to create a clear documentation easy to read by beginners (dummies) like me.
Something like

**1. Read how to create a custom build**
[Building in Windows](https://github.com/iNavFlight/inav/blob/master/docs/development/Building%20in%20Windows.md)

**2. Customize mavlink.c**
Open [inav/src/main/telemetry/mavlink.c](https://github.com/iNavFlight/inav/blob/master/src/main/telemetry/mavlink.c) and add your message between the other messages with this format:
```void mavlinkSendImu(void)

{ 
    mavlink_msg_imu_pack( 
       mavSystemId, mavComponentId, &mavSendMsg, 
       millis(),

       SCALED_IMU(xacc),
       SCALED_IMU(yacc),
       SCALED_IMU(zacc),
       SCALED_IMU(xgyro),
       SCALED_IMU(ygyro), 
       SCALED_IMU(zgyro)
    ); 

    mavlinkSendMessage();
}
```

**2. Customize common.h**
Open [./lib/main/MAVLink/common/common.h](https://github.com/iNavFlight/inav/blob/master/lib/main/MAVLink/common/common.h) and add the code `MAV_DATA_STREAM_SCALED_IMU= ? ? ?` _and where?_

stronnag, 2018-04-04:
Appears dead, closing

closed 2018-04-04 by stronnag

glatorg, 2018-04-04:
Seriously? I'm still unable to find a documentation about it.