iNavFlight/inav #2666: Higher roll pids for autolaunch

madoboy, 2018-01-20:
Hi. Can you add custom or just higher roll pids for autolaunch. It would help the launch and make it easier to recover. But still won't affect mid flying if is to high. Thanks

shellixyz, 2018-01-20:
Hello. I don't think it would be a good solution. What is your issue with launch mode ? If you experience torque issue you can try to increase nav_fw_launch_spinup_time to make the motor accelerate slower and if it is not enough increase nav_fw_launch_idle_thr to reduce the torque experienced by your model even further.

madoboy, 2018-01-20:
Thanks for your reply. Its not about torque roll but if the pids are low to
have smooth flying then when i do autolaunch the servos are to slow to
counter and stabilise the level launch. On my planes are not an issue
because i throw then straight. But on wings throwing from side i have to
many times slow correction. And i tend to not use it because it gonna crash
for sure.

or mute
the thread

shellixyz, 2018-01-20:
What PIFF factors are you using ? They should not have an influence on smoothness. IMO smoothness should be tuned with expo / rates / leveling strengh.

madoboy, 2018-01-20:
Then something is wrong with autotune. When i finished on S800 because the
default values are high and starts to wobble on pitch axis. After that
autolaunch is just impossible for me. Thanks for advices but it's nothing
to do with expo rates when you do autonomous flying with launch and RTH or
similar and you don't have any stick inputs. Maybe my knowledge is not so
big on piff but how i know on pid's are for defining for how hard and long
are the corrections.

shellixyz, 2018-01-20:
PIFF factors should almost only be set depending on aerodynamic properties of your wing. Make sure your elevon throws are not too big. They should be between 15° and 30° as a rule of thumb. I never used autotune but I know it pretty new and not bulletproof. It can also be tuned with the `fw_autotune_*` params and maybe needs to for the S800. If it is wobbling after autotune it means the calculated factors are too high. You should try tuning the PIFF factors manually. To do it you should start with P and I for roll and pitch to 0. Then set the pitch and roll rates to the maximum rates you can achieve in manual unstabilized mode (PASSTHROUGH). After that set the FF factors for pitch and roll so that in ACRO mode when you move the pitch / roll stick to the max the elevons move to 90% of the max travel. Next step is to set the P and I for roll / pitch to a small value. Pitch P/I to 5/5 and roll P/I to 2/2 should be a good start and should not introduce any oscillation. Then go fly in ACRO and increase P only for one axis at a time each time you land until you have oscillations in the corresponding axis. Finally set the P factor to 75% of the value you got. After that you know you have good P and FF factors for pitch and roll and you can increase the I factors for roll / pitch to suit your preferences. Now you can set the pitch / roll rate to you liking. Only after you done all that you can tune the leveling strength and frequency cutoff to adjust the feel in angle mode. If the leveling strength is too high it can also make your wing oscillate.

closed 2018-05-27 by stale[bot]