iNavFlight/inav #2667: Missing sensors with Crossfire and Wing fc

sniperxnl, 2018-01-20:
Guys

i have a wing fc with this firmware
https://github.com/iNavFlight/inav/releases/download/1.8/inav_1.8.0_FF_WINGF4.hex

My Crossfire is activated on uart5 and telemtry activated, i get no sensors like Vbat, Fuel or current when i discover new sensors.

When i take my old omnibus with same crossfire i get all the sensors
GPS, Fuel, Vbat, Current

Any idear guys?

Also switched between crossfire reciever, all have the same firmware so why it work on my omnibus and not on my Wingfc?

sergeysi779, 2018-01-20:
I can't find documentation for this FC. Could you try different UART? Maybe there is a hardware inverter on TX pin of this UART?

shellixyz, 2018-01-20:
I never used crossfire telemetry but it looks like it needs to be inverted so you need an hardware inverter on the UART for it to work. The F3 MCU on the Omnibus FC have inverters integrated on every UART but the F4 MCU doesn't have integrated inverters. FCs with F4 MCU often have only one discrete inverter on one of the UARTs. You probably either need to connect your telemetry on the right UART having the inverter or build an inverter yourself like this one: https://quadmeup.com/smartport-inverter-for-f4-flight-controllers/

sniperxnl, 2018-01-20:
I know from the maker from that WingFc that it have not inverters, i have also quads with crossfire and F4 boards and then it works perfect, uart5 have not inverter so it need to work somehow

digitalentity, 2018-01-21:
FF WING is now released as "FF F35" FC. I have the board on my desk and will try to test in the following days.

sniperxnl, 2018-01-22:
Aah great, 2 guys that i know have the same issue so

furiousfpv, 2018-01-25:
Wrong pin define on UART5
#define UART5_TX_PIN            PC11
=> #define UART5_TX_PIN            PC12. 

We'll change.

digitalentity, 2018-01-25:
Oops, good catch @furiousfpv. Are you doing a PR for the fix?

furiousfpv, 2018-01-26:
Yes, MJ will help me push. Thanks.

closed 2018-05-27 by stale[bot]

proepkes, 2018-09-25:
I have the same issue, using FF35 Lightning FC with iNav 2.0.0 and TBS CrossFire Tx + TBS Crossfire Micro RX

In OpenTX I select "Discover new Sensors" but it doesn't find any. (I've also installed the modchip, FrSky Horus)