iNavFlight/inav #2670: Datachip not found - Matek F405 Mini FC

toddkurtz, 2018-01-21:
Hi, I've entered set blackbox_device = SPIFLASH in CLI and tried enabling/disabling Blackbox in config tab but I continue to get the "no dataflash chip found" error in the top corner of the configurator. I isn't saving BB data either. Can you take a look or provide suggestions?

Thank you!!!

Todd

digitalentity, 2018-01-21:
@toddkurtz this is fixed in the latest code and will be included in the next release (expect one in Feb)

closed 2018-05-27 by stale[bot]

gus7119, 2019-07-31:
Ok I was programming  matek f411 and had the same issue as above. I changed boards to check it wasn't not the board and both of my F411 say No Data Chip found.
Logging is off in config and I can not save any setting if I modify something. Please help it was working fine tested motor and servos as well as flight modes, then it appeared top right.

diogo-f, 2020-04-17:
> Ok I was programming matek f411 and had the same issue as above. I changed boards to check it wasn't not the board and both of my F411 say No Data Chip found.
> Logging is off in config and I can not save any setting if I modify something. Please help it was working fine tested motor and servos as well as flight modes, then it appeared top right.
> Have tried Inav 2.20 and 2.2.1. Happens on both am running latest stable version on f411

Having the same thing with a matek f411. Using betafpv fc with this chip. Did you figure it out?

lwan288, 2023-04-17:
> Having the same thing with a matek f411. Using betafpv fc with this chip. Did you figure it out?

iNav reports the same issue with Matek H743 v2/v3. Not sure how to fix it.

Olivier-OO, 2024-06-02:
I just have the same pb with a MATEK F405-WSE on INAV 7.1.1
It's the same problem with the firmware that i found on MATEKSYS download page and the ones I found on INAV github.
I don't remember if it was too with INAV 6.1.0.
Sorry for my english.

sensei-hacker, 2024-06-02:
> I just have the same pb with a MATEK F405-WSE on INAV 7.1.1 It's the same problem with the firmware that i found on MATEKSYS download page and the ones I found on INAV github. I don't remember if it was too with INAV 6.1.0. Sorry for my english.

The MATEK F405-WSE uses an SD card. It doesn't have flash chip.