iNavFlight/inav #2671: MATEK405 AIO software serial does not appear and working

ekitru, 2018-01-21:
### Board and Version
MATEK405 AIO  https://www.banggood.com/Matek-Systems-BetaFlight-F405-AIO-Flight-Controller-Built-in-PDB-5V-2A-9V-2A-Dual-BEC-for-RC-FPV-Racing-Drone-p-1165338.html?rmmds=myorder&cur_warehouse=CN

INAV/MATEKF405 1.8.0 Nov  1 2017 / 06:50:13 (912d1315)

### Behavior
* On enabling SOFTSERIAL feature not softserial ports can be found at ports tab (also nothing on command *serial*
* enable SOFTSERIAL throught UI or CLI
* appear Softserial ports in Ports tab
* nothing

#### Blackbox logs/Config (.ZIP) where problem occurs
https://pastebin.com/NF2NwqCZ

stronnag, 2018-01-21:
Soft serial is not implemented in this board. This is a feature.

ekitru, 2018-01-21:
I am new in RC and inav and flight controllers.
I see follow in release notes: *Full support of SoftSerial ports on F4-based boards*
Does matek405 is F4 based one?

Soft serial is a program emulation of serial port, it is needed counter and some code for interpretation. 
Does you response means that on Matek405 softserial is not available at all? How can I connect telemetry from r9 to this board? only by using hardware inverter?

stronnag, 2018-01-21:
Regardless of what the release note says, soft serial is not implemented on this board. That is why the board specific documentation does not mention it (and specifically does not describe any pins being allocated for SS). This is due to limited hardware pads exposed on this board.

ekitru, 2018-01-21:
I see, according to matek official materials, softserial is available. http://www.mateksys.com/?portfolio=f405-aio 

As I understood this manual, SS can to allocated to s5 or s6 intead of motor5 or motor6

is it possible to enable SS for this board? or how can I get telemetry on it?

stronnag, 2018-01-21:
Sure, it's enabled in βF, which does not care for all the options iNav offers. This is not betaflight.
You can make you own custom firmware, all the necessary clues are in `target.h` / `target.c`

ekitru, 2018-01-21:
ok, I see. I am new in all it and trying to understand all this stuff. Currently it is quite confusing that feature is present in UI and just not available DE FACTO.

If I will enable it and build by myself firmware, what feature will be broken?

stronnag, 2018-01-21:
the only thing that it breaks will be LED_STRIP and having motors / servos on S5 / S6.

Please can we have this discussion  here OR on slack, not both ...

ekitru, 2018-01-21:
If softserial is not available for matek405 boards, it should be written on release page of feasture should be blocked from enabling

kabturek, 2018-01-22:
ok youre new so this boils down to: matek is not inav, if they have something written on their product page - great but nothing is guaranteed on inav side. Volunteers are working hard to support boards from different manufactures for free. If matek wants to sponsor some development - they should do so. 
It's your task as a user to know what features are supported by your board.

ekitru, 2018-01-22:
Sorry, I read release notes and there was Full support of SoftSerial ports on F4-based boards
I tough that matek405 is based on f4 board.

VvIRONWOLFvV, 2018-04-11:
Softserial is supported but a tab (s5) must be assigned to softserial via cli command in order to see the softserial tab in bf.....Type “resource motor 5 none” in CLI, ENTER,
Type “resource serial_tx 11 A15” in CLI, ENTER
Type “Save”, ENTER
Connect the inverted telemetry signal to S5 pad.......after that you should find softserial in your betaflight tabs

stronnag, 2018-04-11:
The above advice will not work in **iNav**. If you want SS on MATEK F405 boards, you must use the development branch firmware and follow the instructions in the dev branch board documentation.

You can find suitable dev branch hex at  http://seyrsnys.myzen.co.uk/inav_dev/

ekitru, 2018-04-11:
unfortunately iNav does not support dynamic resource allocation as betaflight does.