iNavFlight/inav #2673: Add Battery profiles

giacomo892, 2018-01-22:
I often fly the same plane with LiPO and LiOn. 
What happens is that when I fly with LiOn I can down lot lower with the voltage compared to the LiPo and so battery is reported as critical on the OSD and BEEPER goes off, also this is caused by more sag when on LiOn so I hear the buzzer more often than the needed.

Could be nice to have OSD switchable battery profiles where to specify at least ```vbatmaxcellvoltage``` , ```vbatmincellvoltage``` and ```vbatwarningcellvoltage```  for each profile.

Adding the battery capacity to the profile could be even a better touch allowing current sensor enabled board to fire an alarm when capacity if about to be completely drained. Could even help in the eventuality we add a capacity based failsafe mechanism.

shellixyz, 2018-01-22:
Battery capacity should moved to the battery profiles also because it is already used to calculate remaining battery capacity and "fuel %" exported by telemetry via at least Smartport and probably other protocols.

shellixyz, 2018-01-26:
I'm waiting for #2668 to be merged before making another pull request improving the battery management system and starting working on battery profiles.

Pairan, 2018-02-07:
As I fly 3s LiHV and LiPo with different capacities this would be lovely

closed 2018-05-27 by stale[bot]