iNavFlight/inav #2678: Matek F722-STD  no PPM

gandyman, 2018-01-23:
Using Matek F722-STD board from Bangood with Matek FCHUB-6S.

I use the DEVO-RX717 reciever.
Connect to RX2 - PA3 

Download the software from here http://www.mateksys.com/?portfolio=f722-std#tab-id-5

In Betaflight_3.2.4_MATEKF722  everything works, eg reciever, GPS 

In Inav_1.8.1_MATEKF722 from december and the newest version from Jan. 18  and there is no response on my PPM reciever.

I try now all settings possible, even solder to other UARTS but nothing happen, there is no reaction on sticks or switches.

digitalentity, 2018-01-23:
@gandyman can you try this: [inav_1.8.1_MATEKF722.zip](https://github.com/iNavFlight/inav/files/1655818/inav_1.8.1_MATEKF722.zip)

gandyman, 2018-01-23:
Yes this works like a charme.

Thank you so much.

He wil not arm true a switch now? dont know if that is related to something ?
I have no time atm but wil go on tonight wen I am back.

Thanks again.

digitalentity, 2018-01-23:
@gandyman main screen of Configurator will tell you why arming fails. Hint: all pre-arming checks should be marked by green ticks.

PR #2679 (2018-01-23): Change MATEKF722 PPM in timer
Fixes #2678
closed 2018-01-23 by digitalentity