iNavFlight/inav #2682: Maytek F405 OSD no servo outputs

Merewether, 2018-01-23:
### If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

### Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://www.rcgroups.com/forums/showthread.php?t=2495732), [Slack](https://inavflight.signup.team) or other user support forums & groups (e.g. facebook).

### Board and Version
_Board name goes here. If you have a link where you've purchased the board - please include it_
Matek F405-OSD purchased from getfpv (https://www.getfpv.com/matek-f405-osd-betaflight-fc.html)

_Version of INAV used goes here_
_Use CLI command `version` and put its output here_
INAV/MATEKF405OSD 1.8.0 Nov 1 2017 / 06:51:22 (912d1315)

_If it's a custom-compiled firmware please mention this_

### Behavior
* _Describe the problem_
No servo signals when configured as a flying wing.

* _Steps needed to reproduce the problem_

Load preset "Flying wing Z84"
save / reboot
make sure all green in setup and arm board

* _Expected Results_
Should see PWM signal on S1, S3, and S4

* _Actual Results_
I see PWM signal on S1 only for motor. Servo outputs on S3 and S4 are not present.

_Is this a new feature request?_
no

#### Blackbox logs/Config (.ZIP) where problem occurs
_Go to CLI, execute `dump` command copy its output to [PasteBin](https://pastebin.com) and provide a link to a paste here.
https://pastebin.com/pL0Qzwx9

_Upload a zipped blackbox log (if available)_

stronnag, 2018-01-23:
Does the Servo section in https://github.com/iNavFlight/inav/blob/master/docs/Board%20-%20MatekF405.md help, where is describes how the servo are set up on this board?

Merewether, 2018-01-23:
It states that the servos should be on S2 and S3. There is no output on these pads. Looking at 'resource' on the CLI interface, no servo resources are listed. Looking closer; S2 is C07, S3 is C08, and S4 is C09 which are still all resourced to the motors.

stronnag, 2018-01-23:
look like servos to me:
````
const timerHardware_t timerHardware[USABLE_TIMER_CHANNEL_COUNT] = {
    { TIM5, IO_TAG(PA3),    TIM_Channel_4, 0, IOCFG_AF_PP_PD, GPIO_AF_TIM5, TIM_USE_PPM },

    { TIM3, IO_TAG(PC6),    TIM_Channel_1, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM3, TIM_USE_MC_MOTOR | TIM_USE_MC_SERVO | TIM_USE_FW_MOTOR },
    { TIM8, IO_TAG(PC7),    TIM_Channel_2, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM8, TIM_USE_MC_MOTOR |                    TIM_USE_FW_SERVO },
    { TIM8, IO_TAG(PC8),    TIM_Channel_3, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM8, TIM_USE_MC_MOTOR |                    TIM_USE_FW_SERVO },
    { TIM8, IO_TAG(PC9),    TIM_Channel_4, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM8, TIM_USE_MC_MOTOR |                    TIM_USE_FW_SERVO },

    { TIM2, IO_TAG(PA15),   TIM_Channel_1, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM2, TIM_USE_MC_MOTOR |                    TIM_USE_FW_SERVO | TIM_USE_LED},
    { TIM1, IO_TAG(PA8),    TIM_Channel_1, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM1, TIM_USE_MC_MOTOR |                    TIM_USE_FW_SERVO },

````

Merewether, 2018-01-23:
If I read this correctly, a servo is possible on PC6 (changed for 'Fix MATEK405 servo for tricopter #2610'). But I think that neither PC7 (S2) nor PC8 (S3) have a 'TIM_USE_MC_SERVO' associated with them.

stronnag, 2018-01-23:
That is for 1.8.1, and was reverted (mainly because I didn't want to dismantle my tri)

stronnag, 2018-01-23:
resource does not list the servo on my tri; however it has been flying perfectly for months

Merewether, 2018-01-23:
OK. What can I do to help debug this with my limited tool set (no debugger capability ). I see no PWM signals on any of these ports except for the motor when in fixed wing mode. The board is isolated and I am using an oscilloscope to debug.

stronnag, 2018-01-23:
Have you tried 1.8.1 (development branch) ?

Merewether, 2018-01-24:
Thank you. A pull from the Dev branch worked.

closed 2018-01-24 by Merewether