iNavFlight/inav #2685: Can INAV support STM32F411 series ?

nyway, 2018-01-24:
I have a board,the mcu is STM32F411CEU6.
I chance some codes in target.h and flash it,and nothing happen
So I don't know INAV can support

shellixyz, 2018-01-24:
Too little info. What is it ? There is probably nothing preventing iNav to work on this MCU providing matching target files are filled with the right info.

nyway, 2018-01-25:
@shellixyz 
You mean STM32F411 series can be supported in iNav?
Maybe my target files config error,i will check it

shellixyz, 2018-01-26:
Yes. Where did you get this board from ?

nyway, 2018-01-27:
@shellixyz 
I designed it myself,just for test fly wing and multirotor.
I build target,but not working.I don't know if something wrong in define for f411

shellixyz, 2018-01-27:
Did you put it in the F411_TARGETS group in target.mk ?

```
F411_TARGETS   += $(TARGET)
```

nyway, 2018-01-27:
Yes,same as betaflight .
And everything normal working in betaflight

shellixyz, 2018-01-27:
Well i'm afraid I won't be much more useful. I don't know enough about STM32 MCUs. I guess there is something in the initialization code that blocks the boot-up. I guess you could use HW debug if you broke out the debug pins.

nyway, 2018-01-27:
@shellixyz 
OK, I will continue to test.
Thanks for your help

YFZ450MANNY, 2018-04-05:
I just purchased a FC that has that same F411 MCU. Was hoping to load iNav on it for a delta wing project, but I don’t see a target for it in the flashing dropdown box. :(

digitalentity, 2018-04-10:
@YFZ450MANNY which FC did you purchase? Support for 411 is coming

nyway, 2018-04-10:
@digitalentity DALRCF411 target can support betaflight,but not support in inav yet,i try to change the target.h code ,but not success,could't run.Do you have source code can support F411 now with inav?

fiam, 2018-04-10:
@YFZ450MANNY The problem with that FC specifically for INAV is that it doesn't have any uarts broken out for GPS. Do you plan to solder directly to the MCU?

YFZ450MANNY, 2018-04-11:
@fiam I'm not planning to use GPS at all on this build. Just simple gyro assisted flight and pass-through. It's a very small delta wing. 450mm WS.

digitalentity, 2018-04-11:
@fiam since we added better airplane support to INAV many boards without GPS/compass support suddenly became relevant because of INAV's capability to act as a stabilizer for fixed-wing aircraft (even without RTH and other GPS-assisted features)

fiam, 2018-04-11:
@YFZ450MANNY @digitalentity That makes sense. With GPS modules being so cheap these days, I wouldn't buy an FC without at least one uart, but I understand some pilots just want stabilization.

digitalentity, 2018-06-10:
This is done

closed 2018-06-10 by digitalentity