iNavFlight/inav #2686: Enabling LED_STRIP on Matek F722 causes configurator to hang on connect

MrAlfabet, 2018-01-24:
### Board and Version
Matek FLIGHT CONTROLLER F722-STD
iNav 1.8.1 as listed here: http://www.mateksys.com/?portfolio=f722-std#tab-id-5 24 januari release candidate
INAV/MATEKF722 1.8.1 Jan 24 2018 / 22:00:36 (866170b)

### Behavior
After flashing, I need to reconnect the board, otherwise the configurator will not connect to the board. I don't know if this is significant, but I thought I'd mention it.
The issue is that once I enable the LED strip in the configuration tab and reboot, the configurator (both iNav and BF) will hang and crash chrome when 'connect' is clicked.
* _Steps needed to reproduce the problem_
Flash the firmware listed on the matek site, connect with the configurator, enable LED strip, click save.
* _Expected Results_
You can go on configuring...
* _Actual Results_
Your configurator (and chrome) will hang once you click 'connect'. It will not close (or the whole of chrome including browser will go with it) unless you unplug the board.

_Is this a new feature request?_
No

#### Blackbox logs/Config (.ZIP) where problem occurs
_Go to CLI, execute `dump` command copy its output to [PasteBin](https://pastebin.com) and provide a link to a paste here.
https://pastebin.com/bS54Er16
Everything is default.

Antiheld86, 2018-01-26:
Have the same Issue and already talked to Matek Tech Support. They agreed it is a Firmware Bug and gave it to INAV Dev Team. In Betaflight it works.

digitalentity, 2018-01-28:
Appears to be working fine on latest `development` branch. Hex for testing: [inav_1.8.1_MATEKF722.zip](https://github.com/iNavFlight/inav/files/1670858/inav_1.8.1_MATEKF722.zip)

Antiheld86, 2018-01-28:
Yes, for me its working fine.

digitalentity, 2018-01-28:
Closing (fixed) then.

closed 2018-01-28 by digitalentity

jmillerfo, 2018-01-30:
I can confirm that the development .hex above fixes the LED_STRIP issue.