iNavFlight/inav #2687: SP Racing F4 EVO random or delayed GPS stable hover

VoicOfReason, 2018-01-25:
I ‘m not sure if this is an iNav issue/bug, but I recently switched from the Omnibus F4 board to a SP Racing F4 EVO because the EVO has more UARTs. The Omnibus hovered stable in with GPS modes on on every flight even when satellite count was fairly low. But, with identical hardware and settings, my quad is erratic during GPS hover. Sometimes when I lift off it hovers just fine. Other times, it swill start to drift badly, over correct, and start heading for the ground before I turn off GPS. Today, I let is squire 17 satellites and it all worked perfectly for 25 minutes of flight time including disconnecting battery several times to test. So, does it sound like a lack of # of satellites, or a bug since my hardware is identical with Omnibus setup?

krzysztofmatula, 2018-01-25:
Just to rule out simple things - the mag is operating well, has correct orientation set and is calibrated, right?

VoicOfReason, 2018-01-25:
Yes, east is 90 degrees, and I recalibrated Twice. But of course it also
flies very well after seemingly a longer than normal warmup. If the mag
wasn’t calibrated I would guess that it wouldn’t fly well at all.

VoicOfReason, 2018-01-25:
UPDATE: The same problem occurred after 17 satellite locks so I landed and noticed that in Mobile Flight (iPhone connected via TBS Crossfire) East was no longer facing 90 degrees. So, I re-calibrated the compass and now it flies fine again. So, it appears that every time I fly I have to re-calibrate my compass. I never had to do this before. Is this normal for some people?

VoicOfReason, 2018-01-26:
Is it possible the FC isn't saving the MAG calibration settings long-term?

shellixyz, 2018-01-26:
It's probably not the issue. You can check the mag calibration it is `magzero_x`, `magzero_y` and `magzero_z`.

VoicOfReason, 2018-01-26:
UPDATE #2: I flew today. East was 90 degrees, so compass was fine, I'm guessing, since I also calibrated again. I got up to 17 satellites and took off. Turning on GPS hold, it started flying off at an angle into the sunset. My satellite count stayed high, and I flew to some other locations and kept trying GPS hold but it acted erratic each time. So, I landed, unplugged the battery and waited 5 seconds. Then, I plugged it back in, waited 10 seconds until it was ready to fly again and took off. From this point onward, it hovered with GPS on perfectly, in 40mph winds. So, this seems like a software issue to me. If it was hardware it wouldn't SOMETIMES work. And, it now seems to be fixable by power cycling my drone. 

Any thoughts/ideas? This is a serious problem.

VoicOfReason, 2018-01-26:
Update #3: Went flying again later today after replacing the GPS with a new, different GPS. Calibrated compass. Did the same thing: flew erratic in GPS hold mode with 17 sats until I power cycled and then it flew perfect. 

Also, I noticed that normally I hear a series of 10-15 beeps as GPS is being acquired. I haven't heard my FC do that in the last week.

VoicOfReason, 2018-01-26:
Update #4: Waited 20 minutes and flew again after changing baud rate of GPS. Just as before, GPS hold was erratic at first just like all the other times. I unplugged power for 2 seconds and reconnected and GPS hold was still erratic. So, I landed and unplugged power for 10 seconds. That fixed it. So, it appears that removing power from the system for 10 seconds after initial startup fixes the problem and then it flies as it should. What's going on?

VoicOfReason, 2018-01-27:
FLIGHT #6: Flew today after removing the copper foil beneath the GPS (maybe causing problems??). It behaved exactly as before. It quickly locked onto over 12 satellites. I took off, turned on GPS hold and it started banking and drifting hard like it I was pushing the the pitch forward. I landed and unplugged for 10 seconds, turned back on, flew and turned on GPS hold immediately and it held position perfectly from that point onward. Nothing else changed.

shellixyz, 2018-01-27:
It would probably help to analyze the backbox logs. The mag values, GPS coordinates and number of sats are all recorded.

VoicOfReason, 2018-01-27:
OK. I'll install an SD card the report back soon.

VoicOfReason, 2018-01-28:
Trying to setup SD card blackbox logging. Formatted card with SD Card Formatter but can't get past "fatal SD card error". Anyways....

FLIGHT #7: Today I flew with 17 satellites. It hovered perfectly and did a successful RTH. Then, I flew over to the other side of my yard 15 feet away and turned GPS hold on again. This time it flew up into the air full speed. I quickly turned off GPS hold and descended. I flew out a little ways and hit RTH and it just shot straight up into the air. I turned off RTH and came back and did a test hover. I noticed that when I pointed straight North, it hovered stable but when I starting turning the quad East, it show up into the air at full throttle. So, this is all baffling to me. Like I said everything was working perfectly with the same hardware before I changed flight controllers. Same hardware.

VoicOfReason, 2018-01-28:
FLIGHT #8: Quad has become uncontrollable in GPS modes. It sometimes hovers fine if pointing any direction but straight East. As soon as it points East, it shoots into the sky like a rocket on its own. This includes GPS alt/pos hold modes as well as RTH. 

Still trying to work out SD card issue so can't upload blackbox yet...

krzysztofmatula, 2018-01-28:
"Shooting into the sky" may be caused by accelerometer resonance caused by hard-mounting the controller to the frame (or bad damping). It is clearly visible in the log.

VoicOfReason, 2018-01-28:
The FC is soft mounted and so are all the motors...

I can't view the log yet. I just got it to finally write to the SD card but it only recorded 5 seconds of data and rest of the file is garbage.

VoicOfReason, 2018-01-28:
I've been trying to get a fully working iNav quad working since the summer of 2016. If isn't one thing then it's another. Went outside to finally do a flight with blackbox log recording and now after absolutely no changes to my quad it wobbles out of control (regardless of GPS mode) and falls to the ground........... sigh......

VoicOfReason, 2018-01-29:
OK! Here's the first blackbox log from today. The behavior here is not as terribly erratic as it has been, but when I turned the quad it did start to drift hard and fly off in another direction. I will post more logs this week...

[LOG00002.TXT](https://github.com/iNavFlight/inav/files/1672087/LOG00002.TXT)

VoicOfReason, 2018-01-29:
Is it possible the having 2 MSP ports functioning at once is taxing the FC and causing erratic behavior? I have 1 MSP for DJI Goggles RE (just reads voltage) and another MSP for Mobile Flight. Anyone?

shellixyz, 2018-01-29:
Probably not. And with the F4 MCU even more unlikely. Anyway you can check the CPU usage in the configurator.

VoicOfReason, 2018-01-30:
Yeah, it's only at 15%. I flew again yesterday and tried to do a log but suddenly it's flying perfectly. I don't get it. Here's the erratic behavior summary. 

Sometimes:
1. It flies perfect
2. It shoots to the sky when facing East with GPS hold on
3. It drifts hard and starts to fly away with GPS hold on
4. It wobbles uncontrollably when blackbox refresh rate is set below 100%

I haven't been able to get the erratic behavior on blackbox yet. It's like trying to photograph Big Foot.

Acanevaex, 2018-04-29:
Can you use Mission Planner on a phone connected to quad? You could see if gps position on the map is drifting. Im having position hold issues as well. My 450 with SpF4 Neo and naza proto m8n will hold well for 5 or 10 seconds then start doing circles gradually building speed and spiraling wider and wider out. I am within half mile of cell tower so that may be a factor. 
But try the mission planner or convert your log into the google map gps output. Im still trying to figure that out.

closed 2018-07-12 by stale[bot]