iNavFlight/inav #2688: Multirotor RTH using GPS heading

s0up, 2018-01-25:
Would it be possible to make a RTH feature work on multirotors without compass?  I've been trying to wrap my head around why this is not possible.   I understand it needs to have some groundspeed to calculate heading properly (hence it only being supported in fixed wing) but if the quad were to fly in a similar fashion to a wing with no compass present, I do not see why it would differ greatly. 

Seeing as how finicky compass can be, and how tiny GPS units like the micro ublox that lots of us put on racequads does not include a compass, it seems like this method might be very useful. 

Imaginary Scenario: 

1. RTH Engaged
2. Quad flies to RTH altitude
3. Quad angles pitch forward until GPS heading is calculated
4. Quad begins to correct for heading by continuing to fly forward but slowly yaw to point the nose towards home
5. Continue leaning forwards towards home, correcting yaw as necessary
6. When quad is within X of home, initiate landing

teckel12, 2018-02-02:
@s0up The complexity with a quad is the unknown forward direction.  With fixed wing, it only ever flies head-first.  Quads, however, are just as happy flying backwards or sideways.  Therefore, direction is never known with a quad without a compass.

With that said, I do think it's possible to initiate RTH on a quad without a compass.  It sounds fairly easy on paper, but implementing it would seem to be a challenge.  I think it's one of those things where it would be a real challenge to write the code and it wouldn't work very well, so why not just add a cheap compass instead?

But honestly, a compass is almost always included with a GPS module, they are VERY accurate, simple to connect, cheap, and not at all finicky.  If it's a racing quad, why have a GPS at all?  You're going to crash it long before you lose it and need GPS.

shellixyz, 2018-02-02:
I talked with @s0up about this on the slack channel. He wants this because he would like to use it on a micro quad where there is not much space to put a mag away from interference.

We came to the conclusion that a way to do it would be to pitch the quad to a set angle and then execute a RTH procedure based on GPS heading similar to how it is done on fixed wing. The only condition would be that the pitch angle needs to result in an air speed greater than the expected max wind speed so that we know the forward component of the movement will always be positive.

s0up, 2018-02-02:
Thanks for the replies 👍 I love RTH and I do a lot of long range freestyle, and the micro GPS units I shrink wrap to my battery straps don't have a compass, and even if I use a full size GPS, it is quite noisy.   I would not expect it to work as nicely as a compass, but it would be enough to recover craft in a failsafe or loss of FPV scenario.  I'm fairly sure I would not be alone in wanting something like this.  I love iNav on my wings, and being able to have a similar RTH on a minimalistic quad would be amazing.

@shellixyz brought up a very good point when we were speaking about this, that you could use the drift to calculate the wind direction and move in that direction to get a proper heading as well prior to RTH.

teckel12, 2018-02-02:
@shellixyz Exactly what I was thinking, and that the wind could cause havoc with the logic.  With that said, based upon GPS speed, the pitch could be adjusted to overcome the wind speed (INAV is already doing this to control RTH speed).  Very high-speed wind could cause a failure, but that can already happen even with a compass.  At first, the direction wouldn't be known, so it may very well be going in the wrong direction.  But, as soon as direction is known, it would turn to the correct direction.

That, however, may be the easy part.  The hard part may be once it finds home and needs to lock the location for decent.  Typically, the model will rotate to the same orientation that it was launched from for decent and landing.  This orientation would be removed if there wasn't a compass and it would use the last known direction from the RTH to keep a location lock on decent.  It wouldn't be perfect, and RTH accuracy would be off a bit as a result, but it should basically work.

I'll test some ideas out that may make the entire process fairly trivial.  Heck, even though I have a compass, it would free up a UART for me that I could use for something else.

fiam, 2018-02-02:
There are some implementations of a “simulated” compass using GPS and IMU on multicopter, unfortunately I don’t know of any open source one.

As @teckel12 has pointed out, this is one of those things that are easy to grasp (a quad moves primarily in the orientation it’s tilted to) but pretty difficult to implement with usable results.

If someone’s willing to implement this feature, I see no reason that would prevent the code from being merged into inav.

s0up, 2018-02-02:
@teckel12 if you need any testing feel free to let me know and I'd be happy to help.  I'm in the process of moving this month, but when I get some spare time I will get myself more familiarized with the iNav code and hopefully I can help with the implementation as well if needed.

teckel12, 2018-02-02:
@s0up I'll be sure to let you test it with your quad first ;)

OlivierC-FR, 2018-02-18:
I second this idea, on sub-220mm quad it is very hard to find a place where a mag is not swamped by either the VTX, the 4 motors, 4 ESCs, or the HD camera. Many pilots would love a "basic" mag-less RTH feature, just the quad self-stabilizing, climbing up, and starting to move in a (any) direction and slowly aiming for the home point while it is looking for its real heading, is better than immediately failling to the ground.

dbasch, 2018-02-18:
+1. I've been experimenting with RTH on a 6" with a BN-880. GPS and baro work just fine, but mag is always noisy. I have tried raising it a few cm off the top plate and that doesn't fix things. perhaps because it's still too close to the rx and vtx antennas. I don't really need the quad to land at a specific point, landing along a line would be perfectly fine too. The quad could approach the landing "strip", slow down and land as close as possible to home but not try to correct if it's overshooting.

Even if the quad doesn't land, just coming back and doing pos hold with reasonable accuracy (say 5 meters) is always better than losing a quad because the mag sends it off into a random direction.

teckel12, 2018-02-19:
@dbasch I've had great luck with the BN-880 on 5" and even 3" quads.  Zero problems at all, even with everything packed on a small 3" frame.  I hear many people talk about how bad magnetometers are on quads, but I've never experienced it, not even in the slightest.

With that said, you do need to make sure you have the orientation correct (BN-880 sensor is flipped as well).  With my 5", `align_mag` is set to `CW270FLIP` and with the 3" it's set to `CW90FLIP`.  Once that's done, it's great all the time.  I even use the compass for orientation on my transmitter with a Lua script when flying by sight.

s0up, 2018-02-20:
I removed all of the checks for compass/etc in inav master, and @dbasch tested it on a miniquad without a compass and it works great (and it was super windy)

https://youtu.be/MfvgY_sQ74A

The roll/pitch attitude needs to be factored to correct gps ground course data to prevent weird issues, but this is a start!

dbasch, 2018-02-20:
@teckel12 I've tried everything, and it just doesn't work right. I even raised the magnetometer an inch from the top plate. It's aligned correctly, calibrated outdoors. I really would like to know what's going on, it's possible that it's a bad batch (I've tested four of them, same results) or that you got lucky. Without the magnetometer the quad lands just fine.

ninja-zx11, 2018-03-09:
@dbasch   Not an expert but did you check if your compass readings are changing with lifting or tilting the quad while keeping it pointed towards any direction (eg keeping it pointed towards north)?If yes then you would need to align the compass with flip or something.
My compass (external combined with ublox GPS) was showing proper heading readings but as soon as i tried to lift the quad,the heading readings were going crazy.So after i used CW90 flip,readings are solid.

dbasch, 2018-03-10:
Yes, the compass works just fine when the quad is not armed. It's clearly
magnetic noise from the motors, and I've given up on trying to run a
compass on a 6" miniquad since.

shaiss, 2018-04-23:
So what's the latest here?  I know @dbasch released his bf version with rth. Wheres iNav at as far as implementing something similar?  Thx guys!

WillByrum, 2018-07-19:
Any update on this? Having a micro quad go into "fixed wing mode" for rth would be a big improvement vs trying to get a compass to work properly. Even if the quad just climbed to rth altitude, did the rth and orbited overhead (like fixed wing) would be a big help.

FlyR9, 2019-03-27:
Is there news about this feature? It work very well on betaflight i tested it but the finition is bad (no land and a drop at the home point in the case of a failsafe).

teckel12, 2019-03-28:
@FlyR9 To be fair, it "works" in betaflight, not works very well.  No land and dropping at home point isn't working very well IMHO.  The issue here is that INAV has a higher standard as far as navigation goes.  And the betaflight "solution" while slick and nice to have, isn't exactly up to the INAV standard.

Here's my fear.  A betaflight-like heading is implemented in INAV as a fail-safe for a failed compass.  Then, people assume they don't need a compass at all and we end up with lots of lost and wrecked quads.

I'm not sure how often a compass really fails.  I've never had it happen.  The compass is typically build into the GPS, so if anything, both would fail.  So while we could say this feature would be only for a failed compass, I have a feeling people really want it so they don't need a compass at all, which betaflight is clearly showing we're not there yet.

With that said, I think it's possible, but as betaflight has implemented it.  I think it would need to do more of a fixed wing type RTH, where the quad would return to the home location and then do circles as it descends.  This greatly complicates things but does allow it to land, albeit still not very gracefully and potentially very dangerous.

I think we'll get there, but the betaflight example I believe shows how far away we really are still.

Jetrell, 2019-05-30:
> 
> 
> Any update on this? Having a micro quad go into "fixed wing mode" for rth would be a big improvement vs trying to get a compass to work properly. Even if the quad just climbed to rth altitude, did the rth and orbited overhead (like fixed wing) would be a big help.

Like has been said here. Is there going to be any movement on this project?
I agree with the negatives. 
But now that there are so many 2.5 and 3" quads being released. I think this function would be a very useful addition.

Jetrell, 2019-05-30:
Any update on this? Having a micro quad go into "fixed wing mode" for rth would be a big improvement vs trying to get a compass to work properly. Even if the quad just climbed to rth altitude, did the rth and orbited overhead (like fixed wing) would be a big help.

Like has been said here. Is there going to be any movement on this project?
I agree with the negatives.
But now that there are so many 2.5 and 3" quads being released. I think this function would be a very useful addition. Even if its not perfect.

teckel12, 2019-05-31:
@Jetrell Exactly as I thought, not intended for a failed compass but for a quad without one. I have a 3" quad with compass. The issue here is people not building quads to spec if they want navigation.

Jetrell, 2019-05-31:
@teckel12 its more like.... Wanting to use the functionality of iNav, on a 3" quad. Without having to ditch iNav's RTH ability. 
I'm happy enough to continue using Betaflight with GPS rescue on my 3" units. It works fine for me.  
But iNav has a few extra's that betaflight doesn't... and visa versa.  So i'd like to give them a go on these small quads..... but I'm not going to, if it means that I have to go without a failsafe recovery system.

I have read your earlier posts, with respect to using a mag on a 3" quad. And it doesn't work well on low profile quads, where the Mag is at the same level as the motors. Plus the only GPS/Mag unit that is small enough is the Matek SAM M8Q + 5883. 
And I refuse to place a huge draggy M8N GPS/Mag unit on a 3" quad... its just silly. I like to lay my quads out nicely.... not so they look like a flying circus.

So I don't think the issue is the same for everyone.

teckel12, 2019-05-31:
@Jetrell The HMC5883L and MAG3110 are cheap and very small, even on a breakout board. I'd pay $5-10 for proper RTH rather than a rescue mode where your quad basically hunts you down trying to kill you.

I've seen people activate Bf rescue mode, but always reactivate control.  In a situation like a faulty receiver where rescue mode couldn't be deactivated, you would have a flying killing machine that has marked it's creator as the target. 

With that said, I think maybe there's a safer compromise. I could see a RTH that would then loiter above the home position at a defined altitude. With the goal of it returning home so you could regain control over it (but not land).

Basically, I like the idea of RTH without a compass, I just don't like the landing part, as that could get ugly. If it would RTH then loiter, I could get behind that.

laurentopia, 2019-08-13:
RTH then loiter would be wonderful.

I'm running into BN880 mag not detected and remember that mag setup has always been the one pain point. Either running into technical issues or calibration.

AaronBalint, 2019-09-12:
I have been experimenting with a GPS only navigation solution in betaflight and it is quite feasable.  If you are happy to have a reasonable amount of clear air space around each 'hold position' then the results aren't too bad.  You can take a look at the betaflight project here:
https://github.com/AaronBalint/betaflight_plan

Testing so far has been limited to a Tyro 79 with a micro BN-180.

teckel12, 2019-09-12:
I could see this for fixed wing only. But honestly, for $17 you can get a superior GPS with compass, so I'm not sure it's worth the trouble.

AaronBalint, 2019-09-12:
I think it is worth exploring for the added safety aspect.  Let's say the flight controller detects a continuous and eratic descrepency between the compass and the GPS orientation.  Assuming the GPS data looks consistent (which shouldn't be too hard to verify), the obvious choice is to ditch the compass. Without GPS your toast anyway.  Improving GPS only navigation is surely a benefit.

Dannyallyn, 2020-08-11:
I have been trying to use iNav for 3 weeks now for a work project and I have a $32 Matek GPS/Mag unit and I have since crashed 3 quads and replace somewhere to the tune of $300 worth of hardware because the Mag either doesn't calibrate correctly, ignores the CLI manual offsets that I enter, or decides to change which direction it thinks it north mid flight. I'm really tired of using the compass and it is by far the least reliable piece of hardware on the craft. A mode that uses GPS coordinates to determine heading would be hugely beneficial. It would also reduce hardware costs which is always a plus. let me know how this is going and if I can help in anyway on making this a reality in a polished fashion. I am connected with numerous software engineers who would be more than willing (and even overjoyed) to get to help with the implementation of this idea.

s0up, 2020-08-11:
@Dannyallyn just to add some context, we ultimately ended up adding this feature to BetaFlight as "GPS Rescue Mode" but in a nutshell, GPS error is non gaussian and comes in at too slow of a rate to allow for a RTH algo the way that iNav works.  This complexity is exacerbated by the fact that the craft isn't necessarily always traveling in the direction that the nose is pointing on a multirotor. 

Betaflight GPS rescue mode is much more sloppy fixed-wing style approach that will get the quad near the origin point, but there's a pretty hard limit as to how good such an approach could be (mostly incompatible with iNav, even with some fantastic sensor fusion)

Dannyallyn, 2020-08-11:
Oh okay. Thank you! I watched that video that you posted the link to a while back, and it looked fantastic. I totally get the non-compatibility issue with iNav. I've been slamming my head into the wall for the past week as I was raised on BF.   I really appreciate the reply :)

jettero777, 2020-09-18:
@teckel12
> The HMC5883L and MAG3110 are cheap and very small, even on a breakout board. I'd pay $5-10 for proper RTH rather than a rescue mode where your quad basically hunts you down trying to kill you.
The problem is that MAG requires I2C bus, mini FCs usually don't have it.
Now there are many builds of mini long rangers on FC without I2C and those boards are supported by iNav
>I've seen people activate Bf rescue mode, but always reactivate control. In a situation like a faulty receiver where rescue mode couldn't be deactivated, you would have a flying killing machine that has marked it's creator as the target.
Alternative is to lose the quad over water for example during fail safe.
Anyway such feature would be an option so it will be up to pilot to use it or not.
IMHO better to have such capability as option than to not have any rescue mode if compass is missing/faulty.

lavachemist, 2021-01-17:
> @teckel12
> 
> > The HMC5883L and MAG3110 are cheap and very small, even on a breakout board. I'd pay $5-10 for proper RTH rather than a rescue mode where your quad basically hunts you down trying to kill you.
> 
> The problem is that MAG requires I2C bus, mini FCs usually don't have it.
> Now there are many builds of mini long rangers 3-4" on FCs without I2C bus though those boards are supported by iNav
> 
> > I've seen people activate Bf rescue mode, but always reactivate control. In a situation like a faulty receiver where rescue mode couldn't be deactivated, you would have a flying killing machine that has marked it's creator as the target.
> 
> Alternative is to lose the quad for example over water during fail safe.
> Anyway such feature would be an option so it will be up to pilot to use it or not.
> IMHO better to have such capability as an option than to not have any rescue mode at all if compass is missing/faulty.

You can get a GPS unit like the Matek M8Q-CAN that has a barometer, magnetometer and GPS in one unit. It uses a single UART to connect to the FC via MSP. In that situation, you do not need I2C, and only need one free UART.

ryaneyoung, 2022-12-28:
Any updates on this issue? "Return to overhead" would be fantastic for long range quad where you just go past your limit and need to get it back in range to bring it home. 

I'm new here, but have always been interested in working on something open source.

closed 2023-10-27 by DzikuVx