iNavFlight/inav #2695: 873946d7 appears to break navigation modes

stronnag, 2018-01-26:
Flew  873946d7 today on the precious tricopter
````
# INAV/MATEKF405OSD 1.8.1 Jan 26 2018 / 13:09:40 (873946d7)
````
On engaging PH the machine attempts to fly away. Reverting to `INAV/MATEKF405OSD 1.8.1 Jan 12 2018 / 13:36:33 (55154018)` resumes normal perfect PH behaviour. So something since 2018-01-12 appears to have broken nav modes; I use *appears* as the latest firmware was compiled with gcc 7.2; 55154018 may have compiled with 7.2 or 6.3. 

So unless anyone has some insight into what might have broken this, I'll do some bisect / different compiler tests.

shellixyz, 2018-01-26:
Hmm very odd. Can't see anything obvious in this commit that could break navigation modes. I will test it when the weather allows me to do so.

stronnag, 2018-01-26:
Having found the build for 5515401 in a backup, running strings on the elf shows it was built with 6.3.1; I strongly suspect gcc 7.2, it's not the first time its caused problems on this target.

shellixyz, 2018-01-26:
I'm using GCC 7.2.1 and navigation modes were working for me. I did not test them since last month though.

shellixyz, 2018-01-26:
Also if the `USE_LIGHTS` macro is not defined (and it isn't unless you define it yourself in the target.h file) 873946d changes pretty much nothing to the code.

stronnag, 2018-01-26:
indeed, i'm mainly suspecting it's this board's aversion to 7.2. We'll see tomorrow or Sunday, when I can test more comprehensively and have a laptop in the field on which I can build stuff.

stronnag, 2018-01-28:
arm-gcc 7.3 appeared last night; the F405-OSD exhibits no bad behaviours with 7.3 compiled firmware. So who knows what it was? (compiler, soft-I2C glitch, cosmic rays). Anyway, resolved (and of course I sleep so much better knowing that my FC is less vulnerable to Spectre exploits:).

Closing.

closed 2018-01-28 by stronnag

shellixyz, 2018-01-28:
May be the compiler. I thought I was using 7.2.1 but in reality I am using version 7.1.0

For reference I flew 873946d7 compiled with GCC 7.1.0 today and RTH was working fine.

```
Using built-in specs.
COLLECT_GCC=arm-none-eabi-c++
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/arm-none-eabi/7.1.0/lto-wrapper
Target: arm-none-eabi
Configured with: ../gcc-7.1.0/configure --prefix=/usr --mandir=/usr/share/man --with-pkgversion='Fedora 7.1.0-2.fc26' --with-bugurl=https://bugzilla.redhat.com/ --infodir=/usr/share/info --target=arm-none-eabi --enable-interwork --enable-multilib --with-python-dir=share/arm-none-eabi/gcc-7.1.0/python --with-multilib-list=rmprofile --enable-plugins --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --enable-languages=c,c++ --with-newlib --disable-nls --disable-shared --disable-threads --with-gnu-as --with-gnu-ld --with-gmp --with-mpfr --with-mpc --with-headers=yes --with-system-zlib --with-sysroot=/usr/arm-none-eabi
Thread model: single
gcc version 7.1.0 (Fedora 7.1.0-2.fc26)
```

stronnag, 2018-01-28:
cosmic rays or ropey soft-i2c. Don't care, it's fine now.