iNavFlight/inav #2696: Feature Request free Servo and Motor mix options

Redshifft, 2018-01-26:
Talking to @DzikuVx  about his current work on the servo mixer for the GUI I suggested a simple total assignable option for servos and motors. 
As per nasty picture below servos/motors can be selected from the relevant drop-downs. 
Mix  source or multiple sources are then simply selected for each servo with obviously the option of  direct channel or stabilized FC outputs etc.

This will give Inav the option of  controlling more different flying machine configurations.

Any thoughts? 

[image]

ukmook, 2018-01-27:
Be very handy to have a stabilised servo for altitude. Re: Single rotor / Heli

Redshifft, 2018-01-28:
That picture was just a basic idea for custom aeroplane, if all sources are available as mix masters any manner of flying machine should be able to be catered for. From basic mixing stuff like Elevons, 4 servo wings, differential motor control and thrust vectoring to single or twin rotor Helicopters etc..

closed 2018-05-27 by stale[bot]