iNavFlight/inav #2698: Supporting mixture of servos with different frame rates

flybabo, 2018-01-28:
Is it technically feasible to support multiple servos with different refresh rates such as mixture of analog and digital servos on a fixed wing airframe?
Currently, there's a single global setting for servo refresh rate but servoConfig_t data structure appears to have an individual servoPwmRate for each servo.
Question is if I set different servoPwmRate values for the different servos, is it working accordingly?

shellixyz, 2018-01-28:
Outputs sharing the same timer will output the same frequency. It depends on the FC you are using. Take a look at the target.c file matching the FC you are using to know which outputs can have different frequencies. Other than that it should work.

flybabo, 2018-01-28:
Thanks for the guide. Do we have any cli command to set the servo refresh rate per timer?

shellixyz, 2018-01-28:
Only one PWM frequency for all the servos is supported through the settings you will need to make a custom build.

The PWM frequency is assigned to each servo in the `drivers/pwm_mapping.c` line 284. Just can just set a different frequency there based on the timer index related to the output you want to set the frequency on which is stored in the `timerIndex` variable.

```
pwmServoConfig(timerHardwarePtr, pwmIOConfiguration.servoCount, init->servoPwmRate, init->servoCenterPulse, init->enablePWMOutput);
```

flybabo, 2018-01-29:
Thanks.  I'll extend the cli on my custom build to customize the frame rates.
It will be the first small step to implement the transitional mixers for VTOL :)

closed 2018-05-27 by stale[bot]