iNavFlight/inav #2701: Fixed wing rising vertically in rth or failsafe

Christianmix, 2018-01-29:
Hi, my fixed wing, a fly wing has a problem, I put it on the table and set the pitch slope and leaves the closest to zero, and then it always flies nose down, then I decided to install the new firmware and redo all the configuration, this time I left the pitch 3 degrees up and it flies perfect in stabilized mode, it was magnificent, however in rth or failsafe, it starts to rise vertically with the maximum acceleration, I already leave the option climb before rht -> "never" but it did not work, I do not think it should be 3 degrees more because it goes up vertically and giving a few screws at times. Anyone have any idea what it might be?
.............
my dump

# version
# INAV/OMNIBUSF4PRO 1.8.0 Nov  1 2017 / 06:34:32 (912d1315)

# resources

# mixer
mixer FLYING_WING

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 110 -1 
servo 4 1000 2000 1500 -110 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_UIB
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature MOTOR_STOP
feature GPS
feature RSSI_ADC
feature BLACKBOX
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
[... 481 lines cut ...]
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

#

shellixyz, 2018-01-29:
`set nav_fw_climb_angle = 80` that's why

Christianmix, 2018-01-30:
You are the man!!! I was about to reset everything to start over, it would work, but I would never know what would be wrong. I changed because I thought it was angle that she could climb if necessary and not that she would always use this hehehe rise, and it was like a normal thing, she always does the rth or failsafe using the 80th. Understood? I thought it would be there to be used only when needed, but it was always being used. What would be the default?

Christianmix, 2018-01-30:
found it!!

iNAV’s parameters for fixed wing:

set nav_fw_cruise_thr = 1400 # cruise throttle
set nav_fw_min_thr = 1200 # minimum throttle
set nav_fw_max_thr = 1700 # maximum throttle
set nav_fw_bank_angle = 20
set nav_fw_climb_angle = 20
set nav_fw_dive_angle = 15
set nav_fw_pitch2thr = 10 # pitch to throttle
set nav_fw_roll2pitch = 75 # roll to pitch
set nav_fw_loiter_radius = 5000

closed 2018-02-05 by DzikuVx