iNavFlight/inav #2702: AIR MODE AND RTH DISARM

solmag, 2018-01-30:
### Board and Version
_Beerotor F3 Flight controller from RCTimer.com
_INAV version 1.80

### Behavior
* _I have been flying my quad since last week and never had a problem until someone told me to enable air mode with the arm switch which I did. Unfortunately, I flew out about 3477 ft and activated RTH and my drone disarmed mid-air. I used the GPS coordinates to look for it but did not find it despite the fact that I got to the drop pin on google maps. 
* Should RTH be activated with Air mode?_
* _I expected the air mode to assist the RTH to fly back smoothly
* _Actual Results ended up in a disarm mid-air

mrduncan626, 2018-01-30:
Air mode only allows attitude control of your quad at "0" throttle.  It has nothing to do with RTH.

shellixyz, 2018-01-30:
Air mode is not really that but @mrduncan626 is right it has nothing to do with RTH. Basically air mode allows the control loop to change throttle if needed to make enough correction. It is working as much when throttle is full or near full or when it is 0 or close to 0.

solmag, 2018-01-31:
Thanks for your responses. I still can’t figure out why my drone disarmed when I activated RTH. It was coming back and has travelled about 60 ft when it disarmed. I have flown it out and use RTH and never had any issue. This was my first flight using air mode and my drone got disarmed mid air. Can someone please explain why Inav disarms in flight?

Thanks

shellixyz, 2018-01-31:
iNav only disarm if
- the kill switch was activated
- after the auto disarm delay if it was not receiving signal from your remote for this delay and your throttle was at 0 when it lost signal
- your aircraft lost signal and your failsafe is misconfigured
- the configured failsafe method is `DROP` and your aircraft lost signal

Difficult to say without blackbox logs.

solmag, 2018-01-31:
Yes, it’s difficult to say because I lost the drone, but I have my arm / disarm on a switch and my failsafe was set to RTH. I still had signal when RTH was activated. I’m in the process of uploading the DVR footage to YouTube and will share the link when done.

solmag, 2018-01-31:
This is a screen shot of the drone stat
[image]
it did not show how the drone was disarmed. Usually it says disarm by switch when I disarm after landing.
Also, note that my RSSI was still at 45%

shellixyz, 2018-01-31:
Your OSD screen shot is odd. As you said you should see the "DISARMED BY" line which shows how the FC has been disarmed. Also on the OSD stats screen you shouldn't see the lower parts with Battery voltage / RSSI... It looks like the FC freezed in the middle of the OSD screen refresh. One reason could be because your FC power supply failed.

When you fly a aircraft so far from yourself it is a good idea to have a recording of the GPS coordinates either by recording the FPV feed or through telemetry logs on your remote control or with a computer. Another good idea is to have a locator beacon on your aircraft like a Loc8tor or a tBeacon to help you find it. Otherwise it's often very hard to find it unless you have a pretty good idea of were it landed.

solmag, 2018-01-31:
@shellixyz The screen shot above is from my fatshark DVR. I have the GPS Coordinates on my both the OSD and Taranisa. I have gone there about 3 times now and still haven't found it. Will expand my search to an area where I have not been. After reviewing the video until I lost video signal, the GPS coordinates keep changing while it was dropping. I even took a picture of the GPS coordinate on my Taranis after it crashed.

solmag, 2018-02-01:
This is a screen shot of the quad when it was in RTH Mode before disarming
[image]

digitalentity, 2018-02-04:
Impossible to troubleshoot without a log. Most likely a failsafe setup issue.

solmag, 2018-02-04:
I think the failsafe was set correctly or is there something I have to do in CLI? I checked the RTH box in failsafe and I believe that’s all I have to do. Please correct me if I’m wrong because I’m in the middle of a new build. Thanks

shellixyz, 2018-02-04:
You need to make sure that when you turn off your remote control the FC enters failsafe mode. You can check this in the configurator. When you turn off your remote control a red parachute should appear bellow the battery gauge.

solmag, 2018-02-12:
Just finished building the new drone. Here's a picture of my failsafe tab. RTH was checked. Kindly let me know if I'm missing something.
[image]

shellixyz, 2018-02-12:
Please see my previous comment. This only shows that when the failsafe is activated it will do RTH. You also need to make sure failsafe mode is activated when the receiver loses the radio signal.

Also check the other RTH related settings:
```
set failsafe_throttle_low_delay
set nav_min_rth_distance
set nav_rth_climb_first
set nav_rth_climb_ignore_emerg
set nav_rth_tail_first
set nav_rth_allow_landing
set nav_rth_alt_mode
set nav_rth_abort_threshold
set nav_rth_altitude
```

See this for more info on these settings:
https://github.com/iNavFlight/inav/blob/master/docs/Failsafe.md
https://github.com/iNavFlight/inav/wiki/Navigation-modes#rth-altitude-control-modes
https://github.com/iNavFlight/inav/blob/master/docs/Cli.md

solmag, 2018-02-14:
This is what I have set in my CLI

# set failsafe
failsafe_delay = 5
Allowed range: 0 - 200

failsafe_recovery_delay = 5
Allowed range: 0 - 200

failsafe_off_delay = 200
Allowed range: 0 - 200

failsafe_throttle = 1000
Allowed range: 1000 - 2000

failsafe_throttle_low_delay = 100
Allowed range: 0 - 300

failsafe_procedure = RTH
Allowed values: SET-THR, DROP, RTH, NONE

failsafe_stick_threshold = 50
Allowed range: 0 - 500

failsafe_fw_roll_angle = -200
Allowed range: -800 - 800

failsafe_fw_pitch_angle = 100
Allowed range: -800 - 800

failsafe_fw_yaw_rate = -45
Allowed range: -1000 - 1000

failsafe_min_distance = 0
Allowed range: 0 - 65000

failsafe_min_distance_procedure = RTH
Allowed values: SET-THR, DROP, RTH, NONE

closed 2018-05-27 by stale[bot]