iNavFlight/inav #2707: Inav for Holybro Kakute F4 V2

IgorZaharchuk, 2018-01-31:
Inav for Holybro Kakute F4 V2.
Firmware for testing.

https://www.rcgroups.com/forums/showatt.php?attachmentid=10692676&d=1517341708

IgorZaharchuk, 2018-01-31:
http://www.holybro.com/product/50

digitalentity, 2018-01-31:
@IgorZaharchuk please have a look at #2708 and if your implementation is different - let's combine effort.

digitalentity, 2018-01-31:
Oh, you have source code in your attachment as well. I'll review, thanks.

Azuxul, 2018-02-15:
Link of the first message dosen't work (https://www.rcgroups.com/forums/showatt.php?attachmentid=10692676&d=1517341708) ... 
Please update it.

IgorZaharchuk, 2018-02-16:
1.8 https://www.rcgroups.com/forums/showatt.php?attachmentid=10694258&d=1517380262
1.8.1 
https://www.rcgroups.com/forums/showatt.php?attachmentid=10694843&d=1517412958