iNavFlight/inav #2713: Copy dump Omnibus f4 v2 pro and paste in omnibus f3 osd .....

Christianmix, 2018-02-02:
Guys, I have a fly wing with omnibus f4 v2 pro, and I wanted to put a f3 osd ominibus, can I copy the f4 dump and use f3 or the problem? I did not want to have to redo all the settings, especially the flight modes, I have 8 modes configured, can this be done?

I need to do this because I want to put f4 v2 pro on mini talon.

shellixyz, 2018-02-02:
You can load most of it but you should remove at least
- `acczero_*`, `accgain_*` to avoid forgetting to recalibrate the accelerometer
- `acc_hardware` unless the new board have the same acc/gyro

Christianmix, 2018-02-03:
should I then delete these ???

set acc_hardware = MPU6000
set acczero_x = 65
set acczero_y = -16
set acczero_z = -344
set accgain_x = 4129
set accgain_y = 4079
set accgain_z = 4063

shellixyz, 2018-02-03:
yes

closed 2018-02-08 by DzikuVx