iNavFlight/inav #2714: Reopen Issue #2310 - Servo Throws Restricted In Non-Stabilized Modes

trailx, 2018-02-02:
The default smix for flying wing ailerons is set to 50% for roll and pitch.

This was brought up in issue #2310 and was closed because the poster hadn't flown yet.

In flying my Strix Goblin, I noted that I had very slow response on both pitch and roll with stock flying wing settings.  Most flying wings require a certain amount of throw, and its actually very difficult to adjust this using the inav software.  Smix adjustments are not intuitive or easy to understand, especially given the confusing instructions on the wiki page: https://github.com/iNavFlight/inav/blob/master/docs/Mixer.md

I'd suggest first to change the default smix for flying wings back to 100% mixing from pitch and roll. 
 Who cares if you saturate a control surface when at the extents of throw... you're rarely if ever pulling a maneuver with 100% roll and pitch at the same time.

Second, I would suggest adding something in the GUI for editing smix which would allow easier adjustment and more intuitive mixing.  Alternatively, you can just make the wiki page clearer.  "Painless 360", for instance, made a great video that helped with clarity on this subject, although it is a bit out of date at this point:  https://www.youtube.com/watch?v=qL2LMFF5bJ4

Thanks for your consideration.

DzikuVx, 2018-02-02:
those topics were covered before, and one more time, 50% in flying wing mixer is a correct value.

Please watch: https://www.youtube.com/watch?v=o2RwTeBbvos and https://www.youtube.com/watch?v=VEed3Tgvvt4 . All your questions and suggestions are already covered over there

closed 2018-02-02 by DzikuVx

Redshifft, 2018-02-03:
" I'd suggest first to change the default smix for flying wings back to 100% mixing from pitch and roll.
Who cares if you saturate a control surface when at the extents of throw... you're rarely if ever pulling a maneuver with 100% roll and pitch at the same time."

People are clearly not understanding how to setup Aeroplanes! Forget it has the "autopilot" (yes it is an autopilot and not the required full flight controller of a multirotor) The Aeroplane should fly perfectly in passthrough or even no FC first, long before switching in any stabilised modes for the first time.
To even be considering overdriving the servos indicates something is very wrong - we have not long got the default Elevon mix set to the correct 50%, you might want to remove the default 70% expo (which should not be there for Aeroplanes) and see how much better things feel also.

@DzikuVx  Helpful video for the Plane newbies, should have a link from the INAV Aeroplane setup Wiki :)

trailx, 2018-02-05:
My suggestion was based on passthrough flying.  While my plane flies and controls ok with 50% pitch and roll, I was not getting what I felt was adequate full stick angular changes to the craft attitude based on my experience with other planes.  Yes the 70% expo was also part of the problem - which made it feel extra sluggish.  Expo is at least easy to adjust, smix isn't.

My flying wings are normally set up with 100% contribution from both pitch and roll on the elevons.  In my opinion, clipping the servo output from a 100% roll and 100% pitch input isn't a problem because you will not do that maneuver on a wing.  That's a maneuver that might be performed on a quad - maybe that's where this ideology is coming from.

With a wing, you will either roll 100%, or pitch 100%, but not combine those two inputs in my experience.  So if you're not going to give that particular input, why would you clip the precision of the servo in the range that you spend 99% of your time in (small inputs).  To counteract this imposed handicap, I would have to mechanically change the pinning points on the control horn, and reduce the effective torque of the servo.

Seems that this is more of a mathematical theory of how to properly set up the servo throws so as not to clip their throws based on all possible theoretical inputs.  However it doesn't seem like this theory is based on actual problems from actual flight.  This feels like a "fix" for something that isn't actually broken, in my opinion.

@DzikuVx Thanks for the video links, it does help explain how to use the new smix editor feature.  I agree with @Redshifft, these videos should be located in the wiki.

digitalentity, 2018-02-05:
> clipping the servo output from a 100% roll and 100% pitch input isn't a problem because you will not do that maneuver on a wing

Key word here is "you". The human pilot will not do that maneuver, while FC stabilization code might and will do it if it's required.

trailx, 2018-02-06:
Talk about setting up your plane improperly... I can't imagine a FC outputting that maneuver on a properly set up plane.

Even if the FC demands that maneuver - nothing bad happens.  The plane just responds a little slower than the FC may expect it to because the angular rate of change response won't be linear due to the clipping, but it doesn't hurt anything.

Oh well, seems this ship has sailed.  It would have been good to have two flying wing mixes - one with clipping, and one without.  I have since set my own smix.

For posterity, incase anyone else runs across this, my smix is as follows, to give pitch less throw than roll, which is generally best on a flying wing:

smix 0 3 0 100 0
smix 1 3 1 70 0
smix 2 4 0 -100 0
smix 3 4 1 70 0

digitalentity, 2018-02-06:
A properly setup plane is a plane that is able to execute whatever FC/Pilot might request. 
You can set up your plane as you see fit - this is what `CUSTOMAIRPLANE` mixer is for.

rkolaczek, 2018-02-18:
> smix 0 3 0 100 0
smix 1 3 1 70 0
smix 2 4 0 -100 0
smix 3 4 1 70 0

Such custom mixer means that in some cases during e.g. rth or other autonomous mode servo outputs can exceeded >2000ms or <1000ms ? Or there is min/max (1000ms, 2000ms) limit on the top of mixer rule ?

DzikuVx, 2018-02-19:
There is a mix/max applied by servo min/max