iNavFlight/inav #2720: iNav LED feature does not support AUX mode this is present in Betaflight

jmillerfo, 2018-02-03:
Matek F722-STD
http://www.mateksys.com/?portfolio=f722-std#tab-id-5

iNAV 1.8.1 Jan 28 Matek release Candidate

### Behavior
In Betaflight 3.2.4 individual LEDs can be conditioned on an AUX channel state.  This allows one AUX channel to dim to black all LEDs and another to Turn ON/Off selected (but not all) LEDs.  I use this function to turn ON/OFF Landing Lights while leaving the Nav  and Strobe LEDs on.  This function is a alternative to Throttle (Thrust) and the GUI has a drop down menu that allows selection of throttle, AUX, etc.  If AUX is selected an additional Mode (Mode 7) is generated that specifies the chosen AUX Channel.

iNav does not have this feature but it should be easy to include the latest LED code from Betaflight to produce the desired additional feature.  The LedStrip.MD document lists AUX ON/OFF as a feature early in the document:

     Addressable LED strips can be used to show information from the flight controller system, 
     the current implementation supports the following:

     Up to 32 LEDs.
     Indicators showing pitch/roll stick positions.
     Heading/Orientation lights.
     Flight mode specific color schemes.
     Low battery warning.
     AUX operated on/off switch...
AUX is not listed elsewhere in the document.  Perhaps this was an oversight when the document was imported from BF or CF.

New Request - as far as I can determine.

jmillerfo, 2018-02-04:
I use the *LEDLOW* flight mode to turn OFF/ON all the LEDs but the AUX mode on BetaFlight allows me to also turn OFF a portion of the LEDs while the remainder are left ON.  Check out:
https://github.com/betaflight/betaflight/wiki/LED-Strip-functionality

To affect On/Off I set the Color to Black with White as the next color.  I throw a switch on the transmitter that changes the PWM from 1000 to 2000.  The effect is to have the LEDs turn ON (White) upon toggling the Switch.

closed 2018-05-27 by stale[bot]

Martijn0001, 2021-03-31:
This track has been labeled as inactive, however, I would also appreciate (very much) such a function. I would like to be able to select a different LED pattern depending on an AUX channel. This way I could have:
AUX (6)=1000 LEDLOW (all off)
AUX (6)=1500 Minimal LED, GPS Lock state and Arm state
AUX (6)=2000 Full LED illumination

Being able to use even more "states" would also make sense for a navigation based aircraft. I would love to be able to also define different patterns for when flying in a Leveling mode, acro and navigation mode. I know there is some implementation of LED color based on flightmode, but that is still really limited in configuration options. 

Another thing I would love to be able to do, is to differentiate blinking enabled for Arm State. I would like to have one color LED (Red for on the tail) which is steady on when not armed, and blinks when armed (all Red though). 

Overall really happy with iNav, let me also state that instead of just mentioning those few points that could be even better...