iNavFlight/inav #2722: wing s800 rth not climb why

abdul60, 2018-02-04:
hello when i rth the wing not fly up. but fly down. i use bf3 omnibus and gps ublox7

this my dump inav

Entering CLI Mode, type 'exit' to return, or 'help'

# dump

# version
# INAV/OMNIBUS 1.7.3 Sep  9 2017 / 10:09:29 (974440df)

# resources

# mixer
mixer FLYING_WING

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 125 -1 
servo 4 1000 2000 1500 -125 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature RX_SERIAL
feature MOTOR_STOP
feature GPS
feature CURRENT_METER
feature RSSI_ADC
feature AIRMODE
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
[... 469 lines cut ...]
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 50
set pitch_rate = 50
set yaw_rate = 20
set tpa_rate = 35
set tpa_breakpoint = 1400

#

please help. thanks

shellixyz, 2018-02-04:
Must be because your `nav_fw_cruise_thr` and/or `nav_fw_pitch2thr` are too low. You can also try to reduce `nav_fw_climb_angle` to an angle your wing can manage.

abdul60, 2018-02-05:
clim angle still set 20 
reduce to 10 or 5 ?

and cruise thr and pitchh2thr how many number must i set.

need your help.

shellixyz, 2018-02-05:
`nav_fw_cruise_thr` must be set to the throttle your wing can cruise at.

`nav_fw_pitch2thr` will make the throttle go higher or lower depending if the wing is climbing or descending. If set to 10, climbing 10 degrees and `nav_fw_cruise_thr` = 1500 throttle will be set to 1600. So chose a number enough so that your wing can climb.

You can reduce `nav_fw_climb_angle` to 10 it will only take more time to reach the set RTH altitude

abdul60, 2018-02-05:
okay my friend i will try. thanks alot

abdul60, 2018-02-06:
nice my friend it works.

now.the arrow gps when i go straight to the front the gps arrow point to front why ?

how to.set when i go to the front arrow gps go back

shellixyz, 2018-02-06:
Sorry I don't understand what you mean.

abdul60, 2018-02-06:
im sory for the bad english 

navigation when i go away straight to the west , arrow navigation go west. not go east.

navigation not corect

shellixyz, 2018-02-06:
Are you using a magnetometer (compass) ?

abdul60, 2018-02-06:
no i dont activated. just ublox m7n.

shellixyz, 2018-02-06:
That's odd. I don't know sorry I don't have a plane with OSD. I think the arrow should point to the direction to go home (towards you).

abdul60, 2018-02-06:
maybe i must activated magnometer auto.

i will try

shellixyz, 2018-02-06:
You don't need a magnetometer for wings

digitalentity, 2018-02-08:
You most probably have setup/configuration issue. Lilely both board alignment and servo inversion which by coincidence work correctly in stabilized modes, but not in auto modes. Please verify that model in Configurator reflects motion of the wing.

closed 2018-05-09 by DzikuVx