iNavFlight/inav #2724: Unexpected beeper behavior in a9bf9453

shellixyz, 2018-02-05:
After this revision the beeper makes around 20 new beeps at startup when a fixed wing mixer is selected. I can't find what change makes that happen.

[a9bf9453_fixed_wing_startup_beeps.zip](https://github.com/iNavFlight/inav/files/1693459/a9bf9453_fixed_wing_startup_beeps.zip)

To reproduce: flash firmware then in the configuration tab select a fixed wing mixer (FLYING_WING for example) and enable PWM outputs then `save & reboot`.

giacomo892, 2018-02-05:
I can reproduce it too on OMNIBUS F3 (arm-none-eabi-gcc  7.2.1)  but I don't get anything close to 20 beeps. More like less than 10.

shellixyz, 2018-02-05:
Same compiler version here. It's odd that you don't get the same number of beeps. Did you test with your usual settings or like me with the defaults settings and only flying wing mixer + PWM outputs enabled ? I can be wrong but I have the feeling this bug is going to be hard to track down.

giacomo892, 2018-02-05:
I might have left some other settings on. But beside the number (I might count them in wrong way) you should start to split the commit and see what happens step by step. I'd begin stripping out OSD code and see what happens :)

stronnag, 2018-02-05:
Same here (gcc 7.3). Probably around  extra 10 beeps (vice 0 extra  beeps in normal (TRI) configuration). 

I just changed the mixer type. No OSD on the machine.

shellixyz, 2018-02-05:
@giacomo892 I did that but still can't find where this is coming from.

I did some more tests today and isolated only the restructuring of controlRateConfig_t. Even then it is still doing the beeps. If it's not coming from a compiler bug I don't know since nothing have been changed in the logic, only renames. This still makes the new startup beeps: https://github.com/shellixyz/inav/commit/4879a5ee2e5dcc7d9049315a773ad854cc7634f2

shellixyz, 2018-02-05:
Wait... I just tested a9bf9453 again to be sure it was really coming from fd40892 and it's doing it also now... ☹️

shellixyz, 2018-02-05:
Sorry please ignore my last 2 comments... I did not check the directory I was loading the file from and was not flashing the right file

shellixyz, 2018-02-05:
Found it 😅

PR #2728 (2018-02-05): Fix bug in fd40892d making new beeps at startup
Fixes #2724
closed 2018-02-05 by shellixyz